f4pga/f4pga-arch-defs #1770: Enable FD primitives

acomodi, 2020-11-12:
The latest litex designs do make use of the `FD` primitives which are non other than FDREs with the CE set to one and the R set to 0.

These primitives currently generate errors in the yosys synth step, as it seems that they are not included in the `xilinx/cell_sim.v` shared library:

```
6.2. Analyzing design hierarchy..
ERROR: Module `\FD' referenced in module `\top' in cell `\FD_15' is not part of the design.
Makefile:28: recipe for target 'top.eblif' failed
make: *** [top.eblif] Error 1
```

This is the FD instantiation in the design:
```
FD FD_15(                                                                                                                                                                                                                                                                          
    .C(main_ethphy_clkin),                                                                                                                                                                                                                                                         
    .D(builder_liteethphyrgmii_reset6),                                                                                                                                                                                                                                            
    .Q(builder_liteethphyrgmii_reset7)                                                                                                                                                                                                                                             
); 
```

litghost, 2020-11-12:
> The latest litex designs do make use of the `FD` primitives which are non other than FDREs with the CE set to one and the R set to 0.

This change should be in yosys, as I believe this is an example of a re-targeting.  All re-targeting should be done during synthesis, rather than in P&R.

To confirm this is a re-targetting, in Vivado if you run `create_cell -ref FD test`, I expect it will report being retargetted to `FDRE`/

acomodi, 2020-11-12:
@litghost Thanks and yes, the issue seems to be confirmed a re-targetting one:

```
create_cell -ref FD test
WARNING: [Coretcl 2-1024] Master cell 'FD' is not supported by the current part and has been retargeted to 'FDRE'.
```

acomodi, 2020-11-24:
This has been fixed by techmapping the FD to the FDRE in https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1792.

This needs to be temporary, and a proper fix should be implemented.