f4pga/f4pga-arch-defs #1780: Why the kintex7'dbs exist in prjxray-db but the symbiflow can't use k7's device?

the-centry, 2020-11-17:
I found that the kintex7'dbs exist in prjxray-db,however symbiflow's cmakelist didn't use kintex7? The reason is the loss of primitives in some IP source such as iob18(in the dir of xc/common/primitives)?

litghost, 2020-11-17:
There are couple answers to your questions, but the first part is the easiest.  The reason no kintex7 parts are in symbiflow-arch-defs is those parts were not the initial targets.  The Digilent boards we've been using as platforms are mostly artix7 and zynq7 parts, so that is where initial support was added and tested.

You pointed out correctly that the IOB18 is not currently supported, and that is primarily because IOB18 fuzzing is not in place in prjxray!

So in terms of adding k7 support, you can totally help us move this along!  Steps you can follow to help!

1. Pick a development platform with a kintex7 part
2. Ensure that prjxray has that part emitted.  We are working on a doc for adding parts, PR in progress is here: https://github.com/SymbiFlow/prjxray/pull/1486
3. A blinky design minitest for your development platform (see prjxray/minitests), and verify that all bits are decoded (e.g. `$XRAY_BIT2FASM design.bit > design.fasm && grep unknown design.fasm).
 - 3a. Add fuzzers to solve missing bits (like the IOB18 fuzzers)
4. Standup kintex7 support in arch-defs to reproduce the blinky design!

Once blinky is stood up, you could try more complicated designs.  If you are interested, please let us know!

mithro, 2020-11-17:
The [Digilent Genesys 2 board](https://store.digilentinc.com/genesys-2-kintex-7-fpga-development-board/) would be an excellent first target for Kintex 7 support. The [Xilinx Kintex-7 FPGA KC705 Evaluation Kit](https://www.xilinx.com/products/boards-and-kits/ek-k7-kc705-g.html) would be another good default option to start Kintex 7 work on.

litghost, 2020-11-17:
> The [Digilent Genesys 2 board](https://store.digilentinc.com/genesys-2-kintex-7-fpga-development-board/) would be an excellent first target for Kintex 7 support. The [Xilinx Kintex-7 FPGA KC705 [...]

One downside of Genesys2 is that I believe it isn't part of WebPack?

mithro, 2020-11-17:
BTW - The Digilent Genesys 2 board is $999 USD and the KC705 is $1,695 USD. 

I don't know of any super cheap Kintex-7 boards, but the [Numato Skoll board](https://numato.com/product/skoll-xilinx-kintex-7-usb-ready-to-go-fpga-module/) could potentially work?

the-centry, 2020-11-18:
> There are couple answers to your questions, but the first part is the easiest. The reason no kintex7 parts are in symbiflow-arch-defs is those parts were not the initial targets. The Digilent boards we've been using as platforms are mostly artix7 and zynq7 parts, so that is where initial support was added and tested.
> 
> You pointed out correctly that the IOB18 is not currently supported, and that is primarily because IOB18 fuzzing is not in place in prjxray!
> 
> So in terms of adding k7 support, you can totally help us move this along! Steps you can follow to help!
> 
> 1. Pick a development platform with a kintex7 part
> 2. Ensure that prjxray has that part emitted.  We are working on a doc for adding parts, PR in progress is here: [SymbiFlow/prjxray#1486](https://github.com/SymbiFlow/prjxray/pull/1486)
> 3. A blinky design minitest for your development platform (see prjxray/minitests), and verify that all bits are decoded (e.g. `$XRAY_BIT2FASM design.bit > design.fasm && grep unknown design.fasm`).
> 
> * 3a. Add fuzzers to solve missing bits (like the IOB18 fuzzers)
> 
> 1. Standup kintex7 support in arch-defs to reproduce the blinky design!
> 
> Once a blinky example is stood up, you could try more complicated designs. If you are interested, please let us know!

I noticed that the steps you offered were all involved in prjxray(minitests and fuzzers). Did it mean that adding Kintex7 part into symbiflow only need a complete Kintex 7 database. Do the xml files used by vpr need to be modified to adapt to Kintex7(eg. iob18)?

mithro, 2020-11-18:
@the-centry - There is no point trying to add xml files to symbiflow-arch-defs until Project X-Ray supports them first.

the-centry, 2020-11-18:
> @the-centry - There is no point trying to add xml files to symbiflow-arch-defs until Project X-Ray supports them first.

If  I wanna try whether Kintex7's db could be used in symbiflow firstly, could symbiflow run with the current Kintex7's db?

litghost, 2020-11-18:
> If I wanna try whether Kintex7's db could be used in symbiflow firstly, could symbiflow run with the current Kintex7's db?

It is possible.  I recommend you make a minitest to prove it to yourself.  This is an easy step, and will give you confidence in the current state of the database.

the-centry, 2020-11-18:
> > If I wanna try whether Kintex7's db could be used in symbiflow firstly, could symbiflow run with the current Kintex7's db?
> 
> It is possible. However, I definitely recommend you make a minitest to prove it to yourself. This is an easy step, and will give you confidence in the current state of the databa

the-centry, 2020-11-18:
Follow your guide! I make a IOB18 seg_db,and I use the original minitests/iostandard to test whether is right.but it seems get a wrong design.fasm.Does it mean that I can't use the seg_db in symbiflow? Or I could use it in symbiflow but need to correct the seg_db later?
Here are the content of the design.fasm:
`
# In frame 0x0000001c 2 bits were not converted.
# In frame 0x0000001d 1 bits were not converted.
# In frame 0x0000001e 1 bits were not converted.
# In frame 0x0000001f 3 bits were not converted.
# In frame 0x00000020 3 bits were not converted.
# In frame 0x00000021 3 bits were not converted.
# In frame 0x00000026 9 bits were not converted.
# In frame 0x00000027 28 bits were not converted.
# In frame 0x0000269d 3 bits were not converted.
# In frame 0x0000269e 1 bits were not converted.
# In frame 0x0000269f 3 bits were not converted.
# In frame 0x000026a0 3 bits were not converted.
# In frame 0x000026a1 3 bits were not converted.
# In frame 0x000026a6 27 bits were not converted.
# In frame 0x000026a7 36 bits were not converted.
# In frame 0x0002001c 1 bits were not converted.
# In frame 0x0002001d 2 bits were not converted.
# In frame 0x0002001e 2 bits were not converted.
# In frame 0x0002001f 2 bits were not converted.
# In frame 0x00020020 3 bits were not converted.
# In frame 0x00020021 3 bits were not converted.
# In frame 0x00020026 9 bits were not converted.
# In frame 0x00020027 16 bits were not converted.
# In frame 0x0002269d 3 bits were not converted.
# In frame 0x0002269e 3 bits were not converted.
# In frame 0x0002269f 1 bits were not converted.
# In frame 0x000226a0 3 bits were not converted.
# In frame 0x000226a1 3 bits were not converted.
# In frame 0x000226a6 18 bits were not converted.
# In frame 0x000226a7 15 bits were not converted.
# In frame 0x0004001c 1 bits were not converted.
# In frame 0x0004001d 2 bits were not converted.
# In frame 0x0004001e 2 bits were not converted.
# In frame 0x0004001f 2 bits were not converted.
# In frame 0x00040020 3 bits were not converted.
# In frame 0x00040021 3 bits were not converted.
# In frame 0x00040026 15 bits were not converted.
# In frame 0x00040027 27 bits were not converted.
# In frame 0x0004269c 1 bits were not converted.
# In frame 0x0004269d 2 bits were not converted.
# In frame 0x0004269e 2 bits were not converted.
# In frame 0x0004269f 2 bits were not converted.
# In frame 0x000426a0 3 bits were not converted.
# In frame 0x000426a1 3 bits were not converted.
# In frame 0x000426a6 22 bits were not converted.
# In frame 0x000426a7 29 bits were not converted.
# In frame 0x0006001c 3 bits were not converted.
# In frame 0x0006001e 3 bits were not converted.
# In frame 0x0006001f 1 bits were not converted.
# In frame 0x00060020 3 bits were not converted.
# In frame 0x00060021 3 bits were not converted.
# In frame 0x00060026 16 bits were not converted.
# In frame 0x00060027 12 bits were not converted.
# In frame 0x0006269c 1 bits were not converted.
# In frame 0x0006269d 2 bits were not converted.
# In frame 0x0006269e 2 bits were not converted.
# In frame 0x0006269f 2 bits were not converted.
# In frame 0x000626a0 3 bits were not converted.
# In frame 0x000626a1 3 bits were not converted.
# In frame 0x000626a6 24 bits were not converted.
# In frame 0x000626a7 24 bits were not converted.
# In frame 0x0008001d 3 bits were not converted.
# In frame 0x0008001e 1 bits were not converted.
# In frame 0x0008001f 3 bits were not converted.
# In frame 0x00080020 3 bits were not converted.
# In frame 0x00080021 3 bits were not converted.
# In frame 0x00080026 6 bits were not converted.
[... 1003 lines cut ...]
{ unknown_bit = "004026a6_95_12", unknown_segment = "0x00402680", unknown_segbit = "38_3052" }
{ unknown_bit = "004026a6_96_0", unknown_segment = "0x00402680", unknown_segbit = "38_3072" }
{ unknown_bit = "004026a6_96_2", unknown_segment = "0x00402680", unknown_segbit = "38_3074" }
{ unknown_bit = "004026a7_48_15", unknown_segment = "0x00402680", unknown_segbit = "39_1551" }
{ unknown_bit = "004026a7_49_19", unknown_segment = "0x00402680", unknown_segbit = "39_1587" }
{ unknown_bit = "004026a7_49_31", unknown_segment = "0x00402680", unknown_segbit = "39_1599" }
{ unknown_bit = "004026a7_49_7", unknown_segment = "0x00402680", unknown_segbit = "39_1575" }
{ unknown_bit = "004026a7_59_1", unknown_segment = "0x00402680", unknown_segbit = "39_1889" }
{ unknown_bit = "004026a7_59_23", unknown_segment = "0x00402680", unknown_segbit = "39_1911" }
{ unknown_bit = "004026a7_60_15", unknown_segment = "0x00402680", unknown_segbit = "39_1935" }
{ unknown_bit = "004026a7_60_1", unknown_segment = "0x00402680", unknown_segbit = "39_1921" }
{ unknown_bit = "004026a7_60_23", unknown_segment = "0x00402680", unknown_segbit = "39_1943" }
{ unknown_bit = "004026a7_77_15", unknown_segment = "0x00402680", unknown_segbit = "39_2479" }
{ unknown_bit = "004026a7_78_19", unknown_segment = "0x00402680", unknown_segbit = "39_2515" }
{ unknown_bit = "004026a7_78_31", unknown_segment = "0x00402680", unknown_segbit = "39_2527" }
{ unknown_bit = "004026a7_78_7", unknown_segment = "0x00402680", unknown_segbit = "39_2503" }
{ unknown_bit = "004026a7_87_23", unknown_segment = "0x00402680", unknown_segbit = "39_2807" }
{ unknown_bit = "004026a7_88_15", unknown_segment = "0x00402680", unknown_segbit = "39_2831" }
{ unknown_bit = "004026a7_88_1", unknown_segment = "0x00402680", unknown_segbit = "39_2817" }
{ unknown_bit = "004026a7_88_23", unknown_segment = "0x00402680", unknown_segbit = "39_2839" }
{ unknown_bit = "004026a7_95_23", unknown_segment = "0x00402680", unknown_segbit = "39_3063" }
{ unknown_bit = "004026a7_96_15", unknown_segment = "0x00402680", unknown_segbit = "39_3087" }
{ unknown_bit = "004026a7_96_1", unknown_segment = "0x00402680", unknown_segbit = "39_3073" }
{ unknown_bit = "004026a7_96_23", unknown_segment = "0x00402680", unknown_segbit = "39_3095" }
`

litghost, 2020-11-18:
So that doesn't look too bad.  As expected, there are some missing bits that will need to be investigated.  Based on past experience, most of those are bits are likely from the IO tiles, but you'd need to confirm.  What did you do with the IOB18 tiles?  If you only copied the segbits, that wouldn't be enough.  You need to update `005-tilegrid` to make sure you have base addresses for the IOB18 and IOI18 tiles.

the-centry, 2020-11-19:
> So that doesn't look too bad. As expected, there are some missing bits that will need to be investigated. Based on past experience, most of those are bits are likely from the IO tiles, but you'd [...]

Yea,Through the tilegrid.json genereted by 005,I found that the base addressess of IOB18 exist,but the IOI18 you mentioned did't    exist and it maybe is IOI in tilegrid.json.In my mind, it maybe the problem of 030,because I only changed the 030 to generate IOB18's segbits_iob18.db.Whether the problem is the incomplete of the segbits_iob18.db generated by 030.The second question is that whether I could use this database to run symbiflow and then complets the dababase little by little.

litghost, 2020-11-19:
> Yea,Through the tilegrid.json genereted by 005,I found that the base addressess of IOB18 exist,but the IOI18 you mentioned did't exist and it maybe is IOI in tilegrid.json.In my mind, it maybe the [...]

Please create a PR with your initial changes.  It sounds like a good start, and something that can be built on.

> The second question is that whether I could use this database to run symbiflow and then complets the dababase little by little.

You would need to do some analysis to determine where the missing bits are.  If all of the missing bits are in the IO columns to the left and right of the chip, then you could potential create an ROI the excludes the IO columns.  Creating an ROI is somewhat fragile, and we've pretty much deprecate support for ROI's that have unknown bits in them.

My strong recommendation is to first get your test design fully decoding before you attempt to stand up arch-defs.  This is not required, but will make it much easier to debug what is going on if designs output from arch-defs are not working.

the-centry, 2020-11-20:
> > Yea,Through the tilegrid.json genereted by 005,I found that the base addressess of IOB18 exist,but the IOI18 you mentioned did't exist and it maybe is IOI in tilegrid.json.In my mind, it maybe the problem of 030,because I only changed the 030 to generate IOB18's segbits_iob18.db.Whether the problem is the incomplete of the segbits_iob18.db generated by 030.
> 
> Please create a PR with your initial changes. It sounds like a good start, and something that can be built on.
> 
> > The second question is that whether I could use this database to run symbiflow and then complets the dababase little by little.
> 
> You would need to do some analysis to determine where the missing bits are. If all of the missing bits are in the IO columns to the left and right of the chip, then you could potential create an ROI that excludes the IO columns. Creating an ROI is somewhat fragile, and we've pretty much deprecated support for ROI's that have unknown bits in them. However you could recreate some of that to support your use case.
> 
> My strong recommendation is to first get your test design fully decoding before you attempt to stand up arch-defs. This is not required, but will make it much easier to debug what is going on if designs output from arch-defs are not working.

Ok! Thanks so much! I am very interested in the reason that  why the existed k7'db in prjxray_db don't be used in symbiflow_arch_defs?

the-centry, 2020-11-20:
> Please create a PR with your initial changes. It sounds like a good start, and something that can be built on.
>
>
> You would need to do some analysis to determine where the missing bits are. If all of [...]

OK!Thanks so much!

zjenny09, 2020-11-20:
> Please create a PR with your initial changes. It sounds like a good start, and something that can be built on.
>
>
> You would need to do some analysis to determine where the missing bits are. If all of [...]

I noticed that an roi parameter(design.json) is used when creating synth_tiles.json. Should I create a design.json, if I want to create an ROI?

litghost, 2020-11-20:
Logic for building the harness is located at:https://github.com/SymbiFlow/prjxray/tree/master/minitests/roi_harness

zjenny09, 2020-11-24:
> Logic for building the harness is located at:https://github.com/SymbiFlow/prjxray/tree/master/minitests/roi_harness
> 
> > [image][image] GitHub**[SymbiFlow/prjxray](https://github.com/SymbiFlow/prjxray)**Documenting the Xilinx 7-series bit-stream format. - SymbiFlow/prjxray

When I tried to generate a design.json for kintex7 through minitest, I found that void wires appeared, such as follows:
{ "name": "dout[0]", "node": "INT_R_X23Y95/LH12", "pin": "L25", "type": "out", **_"wire": "", "wires_outside_roi": []_** }, { "name": "dout[1]", "node": "INT_R_X23Y97/LH12", "pin": "M19", "type": "out", **_"wire": "", "wires_outside_roi": []_** } ], "required_features": [ "", "CLK_BUFG_BOT_R_X67Y100.BUFGCTRL.BUFGCTRL_X0Y0.IN_USE", "CLK_BUFG_BOT_R_X67Y100.BUFGCTRL.BUFGCTRL_X0Y0.IS_IGNORE1_INVERTED", "CLK_BUFG_BOT_R_X67Y100.BUFGCTRL.BUFGCTRL_X0Y0.ZINV_CE0", "CLK_BUFG_BOT_R_X67Y100.BUFGCTRL.BUFGCTRL_X0Y0.ZINV_S0", "CLK_BUFG_BOT_R_X67Y100.CLK_BUFG_BUFGCTRL0_I0.CLK_BUFG_IMUX28_0", "CLK_BUFG_BOT_R_X67Y100.CLK_BUFG_BUFGCTRL0_I1.CLK_BUFG_IMUX28_0", "CLK_BUFG_BOT_R_X67Y100.CLK_BUFG_BUFGCTRL1_I0.CLK_BUFG_IMUX29_0", "CLK_BUFG_BOT_R_X67Y100.CLK_BUFG_BUFGCTRL1_I1.CLK_BUFG_IMUX29_0", "CLK_BUFG_BOT_R_X67Y100.CLK_BUFG_BUFGCTRL2_I0.CLK_BUFG_IMUX30_0", "CLK_BUFG_BOT_R_X67Y100.CLK_BUFG_BUFGCTRL2_I1.CLK_BUFG_IMUX30_0", "CLK_BUFG_BOT_R_X67Y100.CLK_BUFG_BUFGCTRL3_I0.CLK_BUFG_IMUX31_0", "CLK_BUFG_BOT_R_X67Y100.CLK_BUFG_BUFGCTRL3_I1.CLK_BUFG_IMUX31_0", "CLK_BUFG_BOT_R_X67Y100.CLK_BUFG_BUFGCTRL4_I0.CLK_BUFG_IMUX28_1",
The void wires made the generation of synth_tiles.json failed. How should I deal with them?

closed 2021-02-26 by the-centry

KKtiandao, 2022-04-14:
@the-centry hi, have you  added iob1.8 and ioi for k7 sucessfully? I saw EXCLUDE_ROI is null in setting.sh on k7.
[image]

mithro, 2022-04-14:
Hopefully interesting links;
 * https://github.com/f4pga/prjxray/pull/1902
 * https://github.com/f4pga/prjxray/issues/1900
 * https://github.com/kazkojima/db-workspace-for-kintex7
 * https://github.com/kintex-chatter

hansfbaier, 2022-04-16:
@the-centry I started implementing an IOB18-fuzzer and we already (with manual hacks) got a blinky (button on IOB18 -> LED on IOB33) running. See this pull request: https://github.com/f4pga/prjxray/pull/1867

maij, 2023-12-22:
I'm not sure why this issue was closed since k7 devices are still not part of the arch-defs yet.
Is there any plan to integrate k7 devices into this flow now that there has been considerable progress in prjxray?

I'm new to this flow, so it's hard to understand how I might do this myself.

hansfbaier, 2023-12-23:
You can use the openXC7 toolchain for Kintex.

the-centry, 2023-12-25:
> I'm not sure why this issue was closed since k7 devices are still not part of the arch-defs yet. Is there any plan to integrate k7 devices into this flow now that there has been considerable progress in prjxray?
> 
> I'm new to this flow, so it's hard to understand how I might do this myself.

According to my tried, there were some wrong data in the timing data and bells.json. And it seems that this flow has been abandoned, scholars research on nextpnr.

the-centry, 2023-12-25:
> @the-centry I started implementing an IOB18-fuzzer and we already (with manual hacks) got a blinky (button on IOB18 -> LED on IOB33) running. See this pull request: [f4pga/prjxray#1867](https://github.com/f4pga/prjxray/pull/1867)

Such a greate work, I am soory for getting the information so late!