f4pga/f4pga-arch-defs #1785: utils/update_tools/update_tools.py needs documentation

HackerFoo, 2020-11-18:
It would be much easier to remember how to use this script (for infrequent users) if there was documentation on how to use it. Something like:

1. Create a fresh checkout:
```
  cd <somewhere>
  git clone <repo>
```
2. Update master in that repo:
```
  git checkout master
  git reset --hard <upstream url>/master
```
3. Run the script:
```
  ./utils/update_tools/update_tools.py --location <somewhere>/<repo> --branch_name master+wip-next
```
4. Push to your fork, then make a PR

litghost, 2020-11-18:
Steps 1 and 2 are not required, you can just point it at a new folder and it does steps 1 and 2 for you.  Your point on documentation is valid.

HackerFoo, 2020-11-18:
Step 1 might not be needed, but step 2 is, or at least rebasing/resetting master from upstream.

litghost, 2020-11-18:
Step 2 should be handled by "pull-bot", e.g. https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/556 The yosys version of pull-bot merges automatically, but the one on SymbiFlow/vtr doesn't.  Regardless, step 2 should be handled in a "public" fashion, rather than a local one.

Step 1 (e.g. clean repo) is the behavior or `update_tools.py` in a clean directory.  I'm wary of having `update_tools.py` doing the cleaning/`rm -rf`, but we could modify it to complain if the directory already has a git repo.