f4pga/f4pga-arch-defs #1788: Input sensitivity status

acomodi, 2020-11-18:
This issue is to keep track of the issue reported here: https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1776.

## Problem statement

The auto-generated designs, such as litex, present some small variations in the memory initialization of the BRAMs. This causes a small perturbation in the initial conditions of a test, which lead to major changes in the output results.

This problem might not be only related to the BRAM initialization, but to any small changes that are applied to a specific design, but for the sake of this issue description, we will keep the small BRAM initialization changes as the triggering factor for this issue.

There are several levels that affect the final output results.

### Synthesis

The mem.init changes make yosys produce two completely different output eblifs. There are some tests that need to be performed in order to asses whether the two different eblifs represent the same circuit, and it is just a matter of ordering.
- run post_synthesis simulations
- apply sorting to the output blifs

The expected output of the synthesis step would be to have only a small number of lines affected, relative to the BRAM initializations.

### Packer

The perturbation in the nets ordering of the circuit might affect the way the packer acts. For instance, two different runs of the same test produced the following packing results:

**Test 1**:
```
Resource usage...
        Netlist
                1099    blocks of type: BLK-TL-SLICEL
        Architecture
                2150    blocks of type: BLK-TL-CLBLL_L
                1200    blocks of type: BLK-TL-CLBLL_R
                1800    blocks of type: BLK-TL-CLBLM_L
                3000    blocks of type: BLK-TL-CLBLM_R
```

**Test 2**:
```
Resource usage...
        Netlist
                1114    blocks of type: BLK-TL-SLICEL
        Architecture
                2150    blocks of type: BLK-TL-CLBLL_L
                1200    blocks of type: BLK-TL-CLBLL_R
                1800    blocks of type: BLK-TL-CLBLM_L
                3000    blocks of type: BLK-TL-CLBLM_R
```

The two Verilog descriptions of the design are identical, except for the memory initialization.

A possible solution to this behaviour is to make the packer algorithms more robust to changes in the ordering of the input circuit, making it less sensitive to changes in input conditions.

### Placer 

Initial placement is currently very sensitive to the seed. In fact, changing the seed can generate very different outcomes, in terms of CPD as well as router run-time.

This is proven by the following test, which, taken two initially equal packed netlists and two different SEEDs, the router behaviour drastically changed:

**default seed**:
```
## Initializing router criticalities took 0.03 seconds (max_rss 3353.3 MiB, delta_rss +0.0 MiB)
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 108: 6 timing startpoints were not constrained during timing analysis
Warning 109: 1521 timing endpoints were not constrained during timing analysis
   1   17.0     0.0    0 2.2e+08    7670   25926   12853 ( 0.439%)  332480 ( 5.2%)   18.939     -70.37     -2.273      0.000      0.000      N/A
   2    5.0     2.8    2 4.9e+07    5694   18419    6520 ( 0.223%)  363815 ( 5.7%)   18.424     -52.74     -1.758      0.000      0.000      N/A
   3    4.1     3.4    4 3.7e+07    4113   13278    4585 ( 0.157%)  382077 ( 5.9%)   18.440     -63.06     -1.774      0.000      0.000      N/A
   4    3.6     4.1    4 3.2e+07    3018   10305    2985 ( 0.102%)  396871 ( 6.2%)   18.349     -49.51     -1.683      0.000      0.000      N/A
   5    3.3     4.9    9 2.8e+07    2131    7832    1730 ( 0.059%)  410556 ( 6.4%)   18.377     -53.28     -1.711      0.000      0.000      N/A
[... 33 lines cut ...]
  15    0.0    30.3    0  104536       2       4       0 ( 0.000%)  450311 ( 7.0%)   19.911     -163.3     -3.245      0.000      0.000       15
```

On a separate test, I have verified and confirmed that having exact same input conditions (seed, packed netlist, etc)  produces the same outputs.

## Steps to reproduce

1. Get the latest version of symbiflow-arch-defs (https://github.com/SymbiFlow/symbiflow-arch-defs/commit/991c74f0e1d579f143878b22d72ab65235f3c18c at the moment of writing)
2. Run one of the litex tests as follows:
```
cd symbiflow-arch-defs
make env
cd build
make minilitex_arty_bit
```
3. Save all the run results somewhere
4. Go to the minilitex_arty build directory and delete the `arty_soc` directory.
```
cd build/xc/xc7/tests/soc/litex/mini && rm -r arty_soc
```
5. Re-run the minilitex test, which will trigger a new litex-design generation:
```
make minilitex_arty_bit
```
6. Compare the newly generated results with the ones saved from the previous run.

acomodi, 2020-11-18:
To temporary work-around the litex differences between different builds, a patch might be applied to litex, as [this line](https://github.com/enjoy-digital/litex/blob/444a605deae6a561dbe2c49bf3062eae6f3cd887/litex/soc/software/bios/main.c#L100) in the bios source code produce the small changes in the mem init file:

litghost, 2020-11-18:
> To temporary work-around the litex differences between different builds, a patch might be applied to litex, as [this [...]

Could we propose a patch to `#ifdef` out that line so we get reproducible builds?  They have other `#define`'s for adding and removing stuff from the BIOS header.

acomodi, 2020-11-18:
@litghost Sure, opened PR here: https://github.com/enjoy-digital/litex/pull/702

mithro, 2020-11-19:
@acomodi Can we split this bug into two seperate things?

 - [ ] (a) Synthesis issue -- Why does a small change in BRAM contents cause anything other than a small change in the output eblif's BRAM init?
 - [ ] (b) Packing issue -- Why do two eblifs which should have almost identical content end up at very different packing results?

It could turn out that due to (a) the assumption that in (b) the two eblifs are actually "almost identical" is incorrect.