f4pga/f4pga-arch-defs #1799: Fresh copy of repo fails on `make env`

HackerFoo, 2020-11-21:
On glinux, from a newly cloned symbiflow-arch-defs repo, I ran `make env` as indicated by the README, and this happens:

```
OS_TYPE=Linux CPU_TYPE=x86_64
'name: symbiflow_arch_def_base' 'symbiflow_arch_def_base'
source /usr/local/google/home/deweese/tmp/symbiflow-arch-defs/env/conda/bin/activate symbiflow_arch_def_base && conda info

     active environment : symbiflow_arch_def_base
    active env location : /usr/local/google/home/deweese/tmp/symbiflow-arch-defs/env/conda/envs/symbiflow_arch_def_base
            shell level : 1
       user config file : /usr/local/google/home/deweese/.condarc
 populated config files : /usr/local/google/home/deweese/tmp/symbiflow-arch-defs/env/conda/.condarc
                          /usr/local/google/home/deweese/.condarc
          conda version : 4.8.3
    conda-build version : not installed
         python version : 3.8.3.final.0
       virtual packages : __cuda=10.2
                          __glibc=2.31
       base environment : /usr/local/google/home/deweese/tmp/symbiflow-arch-defs/env/conda  (writable)
           channel URLs : https://conda.anaconda.org/symbiflow/linux-64
                          https://conda.anaconda.org/symbiflow/noarch
                          https://conda.anaconda.org/pkgw-forge/linux-64
                          https://conda.anaconda.org/pkgw-forge/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/m-labs/linux-64
                          https://conda.anaconda.org/m-labs/noarch
                          https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /usr/local/google/home/deweese/tmp/symbiflow-arch-defs/env/downloads/conda-pkgs
       envs directories : /usr/local/google/home/deweese/tmp/symbiflow-arch-defs/env/conda/envs
                          /usr/local/google/home/deweese/.conda/envs
               platform : linux-64
             user-agent : conda/4.8.3 requests/2.23.0 CPython/3.8.3 Linux/5.7.17-1rodete3-amd64 debian/rodete glibc/2.31
                UID:GID : 298101:89939
             netrc file : None
           offline mode : False

git submodule init
git submodule update --init --recursive
CMake Error at env/conda/envs/symbiflow_arch_def_base/share/cmake-3.18/Modules/CMakeDetermineCXXCompiler.cmake:48 (message):
  Could not find compiler set in environment variable CXX:

  g++-6.

Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "/usr/local/google/home/deweese/tmp/symbiflow-arch-defs/build/CMakeFiles/CMakeOutput.log".
make: *** [Makefile:29: env] Error 1
```

litghost, 2020-11-21:
I believe this is because your CXX variable is set to `g++-6`?  If you unset CXX and CC, it will work?  Recently gLinux removed old compiler versions.  I got bit because clang-7 no longer existed.

mithro, 2020-11-21:
Does symbiflow-arch-defs actually require a C++ compiler?

litghost, 2020-11-21:
> Does symbiflow-arch-defs actually require a C++ compiler?

Yes, bitstream tools et/all?

HackerFoo, 2020-11-21:
I think `conda init bash` and restarting the shell fixes this.

At least, I did that, and `make env` successfully completes now.

mithro, 2020-11-21:
I'm going to leave this open and look at it a bit more closely...

HackerFoo, 2020-11-21:
At least the README needs updated.