f4pga/f4pga-arch-defs #1800: Missing support for xc7a35tftg256-1 in archive

dnltz, 2020-11-21:
Hi,

I downloaded the toolchain as guided in the symbiflow-example README file and wanted to try the xc7 counter example for the xc7a35tftg256-1 part. I saw there is support for it in prjxray-db but unfortunately this part is not deployed to the tar archive.

Error message from the symbiflow_place:
```
cd build && symbiflow_place -e top.eblif -d xc7a50t_test -p /home/daniel/tmp/counter/arty.pcf -n top.net -P xc7a35tftg256-1 -s /home/daniel/tmp/counter/counter.sdc 2>&1 > /dev/null
realpath: /home/daniel/tmp/symbiflow/xc7/install/bin/../share/symbiflow/arch/xc7a50t_test/xc7a35tftg256-1/pinmap.csv: No such file or directory
make: *** [Makefile:26: build/top.place] Error 1
```

Seems like only the parts from the two development kits get deployed. Since I have a custom board, can I generate this pinmap by myself?

tcal-x, 2020-11-22:
Hi @dnltz , 

There is a guide for this case -- https://prjxray.readthedocs.io/en/latest/db_dev_process/newpart.html.   You would just be adding a new package (256 pins) for a currently supported device (35t).    So you'd skip steps 2-4 of that guide.  

_However_, we have planned a project to automatically generate the pinmaps for all packages once a device is supported -- https://github.com/SymbiFlow/prjxray/issues/1476 -- but I don't know what the timeline is for that.

dnltz, 2020-11-22:
Thanks @tcal-x ! That was exactly what I was looking for :)

Yes, would be nice to have all available packages in the example. Do you need more hardware for testing? I could send a board with this package in January.

tcal-x, 2020-11-23:
Hi @dnltz ,
I don't we'll need real hardware, although thank you for the offer.   We test mainly by running designs through Symbiflow, then exporting back into Verilog+constriants, reading that into Vivado, then generating a bitstream from there, and diffing between the bitstreams.   This is done in the symbiflow-arch-defs repository.   So we'll want to add some tests there.   Here for example is my PR for adding tests for the 100T:  https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1515/files.   You would touch just a few files  -- a new entry in the boards file xc/xc7/boards.cmake, then some new test variants under xc/xc7/tests/.   Unfortunately I haven't yet written the step-by-step guide for this side of things.

litghost, 2020-11-30:
@dnltz We would like someone to do https://github.com/SymbiFlow/prjxray/issues/1475 and https://github.com/SymbiFlow/prjxray/issues/1476.  It should just be some Python work, if you feel up for it!

dnltz, 2020-12-09:
hey @litghost - I'm very busy right now until end of December. I can do these PRs than, if it's not that urgent.

dnltz, 2021-03-19:
Coming back to my original problem, the way to make my part available in the prebuild toolchain is to add it as board to this project, right?

litghost, 2021-03-19:
> Coming back to my original problem, the way to make my part available in the prebuild toolchain is to add it as board to this project, right?

Next step is to update the prjxray-db package. I'll poke @mithro

mithro, 2021-03-19:
The Zynq build got stuck in some way, so I killed it and kicked it off again.

In the meantime, I pushed the Artix + Kintex stuff to https://github.com/SymbiFlow/prjxray-db/compare/master...mithro:master

There seems to be a lot of changes in the `mask_bram_l` and `mask_dsp_l` files -- is that expected?

mithro, 2021-03-20:
A zynq run completed successfully, see the changes at https://github.com/SymbiFlow/prjxray-db/compare/master...mithro:master

If someone confirms that the mask changes for `bram` and `dsp` are expected, I'll push the result to the database.

acomodi, 2021-03-22:
I am not sure what are the mask files used at this point, or if they are every used in the flow. As for they are expected or not, the only recent changes to the DSP happened [here](https://github.com/SymbiFlow/prjxray/pull/1591) and the output is already checked-in in the db, so the change seems a bit odd.

dnltz, 2021-03-30:
I think I can close this issue. @mithro Can I add a not for sale board with this part?

mithro, 2021-03-30:
@dnltz - Add to were?

dnltz, 2021-03-31:
@mithro - My understand is that the flow needs data from `xc7a50t_test` which are generated with this project. So my idea was to add the board here to this project to have my part deployed. Not sure if this is the right way.

mithro, 2021-03-31:
@dnltz Oh, I understand. It might be worth chatting with @litghost if there is a good way to support all the parts now? If there isn't an easy way, landing that option in the short term is 100% fine.

litghost, 2021-04-01:
@dnltz So you remember how we made the mapping from part -> device -> fabric?  That same distinction exists in arch-defs, but pre-dates the prjxray change.  arch-defs simple maps part directly to fabric (and a "board") in https://github.com/SymbiFlow/symbiflow-arch-defs/blob/a3c75b4b7304a29ff3d48b594e5afe922ce00f7e/xc/xc7/boards.cmake

The key CMake function is `add_xc_board`, which is defined here: https://github.com/SymbiFlow/symbiflow-arch-defs/blob/a3c75b4b7304a29ff3d48b594e5afe922ce00f7e/xc/common/cmake/device_define.cmake#L1-L11

The key portion for expanding part support for everything is generating "pinmap.csv" for every part in prjxray, which is here: https://github.com/SymbiFlow/symbiflow-arch-defs/blob/a3c75b4b7304a29ff3d48b594e5afe922ce00f7e/xc/common/cmake/device_define.cmake#L160-L172

In summary:
 - arch-defs manually mapped part -> fabric in `xc7/boards.cmake`
 - That information now exists in prjxray per your update
 - The only thing that needs to be done to support more parts is to create the `pinmap.csv` for each part.  In order to generalize this, there needs to be a map from prjxray fabric to arch-def `DEVICE`.
   -  For example, the prjxray fabric "xc7a50t" is the arch-def `DEVICE` "xc7a50t".

Because the naming conventions in arch-defs appear to generally match the naming conventions from prjxray, I believe it should be possible to emit CMake targets to generate the "pinmap.csv" for every part now in prjxray as part of the install targets.

Does this make sense?

litghost, 2021-04-01:
One additional note, the current `add_xc_board` names the output pinmap CSV based on the name of the board, but it probably should just be named for the package.

dnltz, 2021-04-07:
I'm not very familiar with cmake. A target is a function like `ADD_XC_BOARD`, right? If that's true, my understanding or idea would be to add a new function to generate all pinmaps for a fabric and to add the function calls to the `boards.cmake` for each device.

dnltz, 2021-04-12:
> I'm not very familiar with cmake. A target is a function like `ADD_XC_BOARD`, right? If that's true, my understanding or idea would be to add a new function to generate all pinmaps for a fabric and to add the function calls to the `boards.cmake` for each device.

@litghost