f4pga/f4pga-arch-defs #1829: CI on master does not run properly

acomodi, 2020-12-02:
While on PRs Kokoro CI works correctly, there seem to be an issue when PRs are merged and CI should start on master.

The problem is that on the latest merges (e.g. https://github.com/SymbiFlow/symbiflow-arch-defs/commit/a6eb1c1d8cb450ca30566e6d571b51034e237dd7 or https://github.com/SymbiFlow/symbiflow-arch-defs/commit/ff71ca792f7620a6d61558056e4565c01bed3f91), Kokoro seem to be stuck.

litghost, 2020-12-02:
Duplicate of https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1826

closed 2020-12-02 by litghost

acomodi, 2020-12-03:
@litghost All right. One thing I do not understand though is that, if kokoro CI do fail on master, they should still show up being red, so to inspect the log files, etc. I am unsure why it is not happening in this case (e.g. in this merge commit https://github.com/SymbiFlow/symbiflow-arch-defs/commit/a6eb1c1d8cb450ca30566e6d571b51034e237dd7 only the GH actions red CI is listed)