f4pga/f4pga-arch-defs #1846: Prune submodules and move packages to PyPi or conda.

litghost, 2020-12-04:
There are many submodules that right now only exist to be installed into the local python environment or to build binaries that exist in conda.  Many can likely be changed to simply use the PyPi version or conda package once ready.

duck2, 2021-01-22:
I may give this a try. Do we only need to change submodule paths in requirements.txt to PyPI packages?

litghost, 2021-01-22:
> I may give this a try. Do we only need to change submodule paths in requirements.txt to PyPI packages?

Depends.  I think for some of them, we need to push the packages to PyPI, which is a little more work.

litghost, 2021-01-28:
Packages that should move to PyPi:
 - https://github.com/SymbiFlow/symbiflow-rr-graph
 - https://github.com/SymbiFlow/python-fpga-interchange
 - https://github.com/SymbiFlow/fasm
 - https://github.com/SymbiFlow/xc-fasm
 - https://github.com/SymbiFlow/python-sdf-timing
 - https://github.com/SymbiFlow/python-symbiflow-v2x
 - https://github.com/SymbiFlow/vtr-xml-utils
 - https://github.com/SymbiFlow/symbiflow-xc-fasm2bels

mithro, 2021-01-28:
Documentation is at https://packaging.python.org/guides/publishing-package-distribution-releases-using-github-actions-ci-cd-workflows/

mithro, 2021-01-28:
`secrets.PYPI_TEST_PASSWORD` should be available on any public repository.

mithro, 2021-01-28:
Once we have it working, `secrets.PYPI_PASSWORD` can be used to publish to the real PyPi.