f4pga/f4pga-arch-defs #1864: make all_demos

nlwmode, 2020-12-09:
The ERROR:
---
OSError: Unable to find any of the cross compilation toolchains:
- riscv64-unknown-elf
- riscv64-unknown-linux-gnu
- riscv32-unknown-elf
- riscv32-unknown-linux-gnu
- riscv64-elf
- riscv32-elf
- riscv-none-embed
- riscv64-linux
- riscv64-linux-gnu
- riscv-sifive-elf
- riscv64-none-elf
xc/xc7/tests/soc/litex/mini/CMakeFiles/file_build_xc_xc7_tests_soc_litex_mini_arty_a7_soc_gateware_top.pcf.dir/build.make:81: recipe for target 'xc/xc7/tests/soc/litex/mini/arty_a7_soc/gateware/top.v' failed
make[3]: *** [xc/xc7/tests/soc/litex/mini/arty_a7_soc/gateware/top.v] Error 1
CMakeFiles/Makefile2:214025: recipe for target 'xc/xc7/tests/soc/litex/mini/CMakeFiles/file_build_xc_xc7_tests_soc_litex_mini_arty_a7_soc_gateware_top.pcf.dir/all' failed
make[2]: *** [xc/xc7/tests/soc/litex/mini/CMakeFiles/file_build_xc_xc7_tests_soc_litex_mini_arty_a7_soc_gateware_top.pcf.dir/all] Error 2
CMakeFiles/Makefile2:18648: recipe for target 'CMakeFiles/all_demos.dir/rule' failed
make[1]: *** [CMakeFiles/all_demos.dir/rule] Error 2
Makefile:454: recipe for target 'all_demos' failed
make: *** [all_demos] Error 2
---
Q:
---
ubuntu 16.04/18.04 , I all tried this two OS.
And if anybody know how to solve this problem?
---

acomodi, 2020-12-09:
Hi @nlwmode, thanks for filing this issue.

I have tried to reproduce the error, but I couldn't, and I did not get the error you reported.

The `riscv64-unknown-elf` package is used to generate the LiteX bios and it is downloaded and installed through conda, which hints to the fact that maybe there were some issues during the environment build.

What I have done with a freshly cloned repo is the following:
```
make env
cd build && make all_demos
```

The above commands worked on my machine (which has Ubuntu 18.04).

Can you please provide all the steps you performed?

nlwmode, 2020-12-20:
My steps:
1. git clone https://github.com/SymbiFlow/symbiflow-arch-defs.git
2. cd symbiflow-arch-defs
3. make env
4. cd build && make all_demos

litghost, 2021-01-05:
@nlwmode Try this:
```
git clone https://github.com/SymbiFlow/symbiflow-arch-defs.git
cd symbiflow-arch-defs
make env
make enter
cd build && make all_demos
```