f4pga/f4pga-arch-defs #1884: Kokoro running out of memory

HackerFoo, 2020-12-15:
Tests are failing on Kokoro with:

```
virtual memory exhausted: Cannot allocate memory
```

e.g. https://source.cloud.google.com/results/invocations/c40e6db8-e047-466f-87c4-483eb97d6429/targets/foss-fpga-tools%2Fsymbiflow-arch-defs%2Fpresubmit%2Fdocs/log

litghost, 2020-12-15:
> Tests are failing on Kokoro with:
> 
> ```
> virtual memory exhausted: Cannot allocate memory
> ```
> 
> e.g. https://source.cloud.google.com/results/invocations/c40e6db8-e047-466f-87c4-483eb97d6429/targets/foss-fpga-tools%2Fsymbiflow-arch-defs%2Fpresubmit%2Fdocs/log

Why did you file an issue that is a result of a change in your own PR in #1880 ?

HackerFoo, 2020-12-15:
It only fails on some of the builds, so it appears to be a CI issue.

litghost, 2020-12-15:
> It only fails on some of the builds, so it appears to be a CI issue.

Those VM machine are smaller, but the issue is that you added building of ANTLR4 to jobs like the `docs` jobs, etc.  Again, this issue arose in your PR only, so I'm not sure why you've filled an issue.

closed 2020-12-18 by HackerFoo