f4pga/f4pga-arch-defs #1902: Mechanism to define pin-mapping for embedded FPGA (eFPGA)

lnsharma, 2020-12-18:
**Problem Statement**:
In eFPGA device, IO interface consists of input/output buffers (registered/non-registered) unlike fixed FPGA which has inpad/outpad connected to some kind of GPIO. Each IO interface in an eFPGA can have multiple such input/output buffers. User of an eFPGA need to then define a mapping between IO buffers they want to connect with other SOC interfaces that communicate with eFPGA (e.g. Wishbone, APB, SPI. GPIO etc.). We also need a mechanism to define clocks & timing budget for selected IO buffers as each SOC interface may have different timing requirement.  This is somewhat similar to custom package definition for fixed FPGA devices. In Symbiflow, we need to define a mechanism to create an eFPGA device with custom IO specification and defining timing budget.

**Description**:
In QuickLogic eFPGA devices, we have IO interface buffers with port names like F2A (where F2A stands for Fabric to ASSP i.e. output port from the device) or A2F (where A2F stands for ASSP to Fabric i.e. input port to the device). Each IO interface cell (at a specific row/column) can have multiple such ports.
User of such eFPGA, then have a choice to assign IOs based on their applications  and system level interconnection to the eFPGA.
We need to provide a way to get this mapping information from the user. This can be done by taking this mapping information in an xml file or a csv file or any other file.  This mapping information can be @processed somewhat similar to how Symbiflow processes package information for FPGA with
Pin mapping csv file.
The end-user of this eFPGA device should also be able to specify fix placement constraints on these custom pin names. For defining timing budget, user can define it in sdc file (preferably) or in the same csv or xml file where pin mapping is defined.

**Proposal**:
Attached document explains pin mapping file specification. If we are defining a sdc file for timing budget then the clock related attributes can be removed from the proposed xml file.
In Symbiflow, this xml file is read in along with architecture definition xml. Internally, this is processed and pin-mapping information is saved in the similar manner as package information is saved for FPGA.

[Pin Mapping File Specifications.docx](https://github.com/SymbiFlow/symbiflow-arch-defs/files/5714450/Pin.Mapping.File.Specifications.docx)

@mithro , @kgugala , @mkurc-ant : please go through this proposal and share your views/thoughts on how can we address pin-mapping for eFPGA devices in Symbiflow.

mithro, 2020-12-18:
@lnsharma - Can you move the "Pin Mapping File Specifications" document to Google Docs so we can do interactive collaboration on it?

mithro, 2020-12-18:
@lnsharma - It would be good to figure out how this interacts with SDC which is what we are moving to for all types of timing and IO constraints.

lnsharma, 2020-12-21:
> @lnsharma - Can you move the "Pin Mapping File Specifications" document to Google Docs so we can do interactive collaboration on it?

ok, I changed the URL and document link now points to google docs.

lnsharma, 2020-12-21:
> @lnsharma - It would be good to figure out how this interacts with SDC which is what we are moving to for all types of timing and IO constraints.

I updated the doc with a section on sdc commands needed for this purpose.