f4pga/f4pga-arch-defs #1925: Vivado diff FASM fails with DRC PLCR-1

mkurc-ant, 2020-12-31:
```
ERROR: [DRC PLCR-1] Placement Constraints Check for Clock Region(s): Design Check found an error in Clock Region X0Y0. This clock region has 17 clocks locked whereas only 12 clocks can be routed per clock region. The cells were either constrained through Area Groups or locked via user constraints. This situation can be resolved by unlocking some cells from the user constraints so that there are only 12 clocks in this region.
```

This happens for the `mmcm_ext_basys3` test under `xc/xc7/tests/mmcm` (for now in https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1729 PR). The test utilizes a `MMCM` with external clock feedback which drives 6 counters.

The surprising thing is that Vivado does not report any routing net conflicts, the error is thrown during bitstream generation phase.

litghost, 2021-01-05:
This is basically the same error as https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1346

Basically, the explicit BUFH that fasm2bels emits should instead be changes to route-thrus, if `BUFHCE.CE = 1`.

mkurc-ant, 2021-08-23:
https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1729 has been merged. Closing.