f4pga/f4pga-arch-defs #1959: How to add IOB18 into symbiflow?

the-centry, 2021-01-20:
#https://github.com/the-centry/symbiflow_problem/issues/1
if I want to add IOB18 into the channels.db and use it later ,whether the step I tried is right and what should I try?

litghost, 2021-01-20:
The first step is to make sure that the IOB18 fuzzers are in place.  We recommend you write a minitest via Vivado that exercises the IOB18 tile in your desired configuration, and ensure there are no unknown bits.

the-centry, 2021-01-21:
> The first step is to make sure that the IOB18 prjxray fuzzers are in place. We recommend you write a minitest via Vivado that exercises the IOB18 tile in your desired configuration, and ensure there are no unknown bits. Please open a PR on https://github.com/SymbiFlow/prjxray/ with your minitest so it can be added to the tests.

It seems beyond my ability.Currently,I only try to  generate the tile's bit that I want to get by modifying your code. And before add it to symbiflow,I have sucessfully in generating the design.json in minitest's roi_harness with the IOB18'db.

closed 2021-02-24 by the-centry