f4pga/f4pga-arch-defs #1973: symbiflow_write_xml_rr_graph toolchain wrapper is not installed

Xiretza, 2021-01-25:
Needs to be added here: https://github.com/SymbiFlow/symbiflow-arch-defs/blob/bff52005b478381125b235c66f394f0f993db287/xc/common/cmake/install.cmake#L26-L34

litghost, 2021-01-25:
While I don't disagree that the wrapper is not installed, I'm unclear why you want it?  It's typically just usefully for developers debugging the RR graph?

Xiretza, 2021-01-25:
I just noticed this when trying to apply my in-tree patches (including to write_xml_rr_graph) to the prebuilt tarball (which doesn't have it). Since it was the only wrapper that was missing, I assumed it was an oversight - if that's not the case, this can be closed.

litghost, 2021-01-25:
I don't believe it is an oversight.  This wrapper is only really useful for arch-def developers.  I don't believe an end user would need it.  Closing for now.  @acomodi Any other thoughts?

closed 2021-01-25 by litghost

acomodi, 2021-01-25:
> Any other thoughts?

@litghost No, I agree that this should not be required/useful for the end-user