f4pga/f4pga-arch-defs #1991: PLLE2_BASE techmap doesn't match yosys' cells

Xiretza, 2021-01-29:
yosys [`cells_xtra.v`](https://raw.githubusercontent.com/YosysHQ/yosys/master/techlibs/xilinx/cells_xtra.v) contains:

```verilog
    output CLKFBOUT;
    output CLKOUT0;
    output CLKOUT1;
    output CLKOUT2;
    output CLKOUT3;
    output CLKOUT4;
    output CLKOUT5;
    output LOCKED;
    input CLKFBIN;
    input CLKIN1;
    input PWRDWN;
    input RST;
```

while [`cells_map.v`](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc/xc7/techmap/cells_map.v#L4295) contains:

```verilog
input         CLKFBIN,
input         CLKIN,

output        CLKFBOUT,
output        CLKOUT0,
output        CLKOUT1,
output        CLKOUT2,
output        CLKOUT3,
output        CLKOUT4,
output        CLKOUT5,

input         RST,
output        LOCKED
```

Compared to yosys, `CLKIN` should be `CLKIN1`, and the `PWRDWN` pin is missing - fixing these two makes techmap work. This is the only module I had problems with, I don't know if something similar applies to other modules.

litghost, 2021-01-29:
Per https://github.com/Xilinx/XilinxUnisimLibrary/blob/1c8e05fd1e9a79ceb8b996a0996674122eed086f/verilog/src/unisims/PLLE2_BASE.v#L37-L48 , I agree that the techmap in https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc/xc7/techmap/cells_map.v#L4295 is wrong.

litghost, 2021-01-29:
@mkurc-ant Can you fix this when you bring in the MMCM support?

mkurc-ant, 2021-02-01:
@litghost Sure, I'll take a look into it.