f4pga/f4pga-arch-defs #2032: efinix FPGAs?

eljuligallego, 2021-02-16:
Hello,
I have tested the Efinix IDE and realize when compile some projects they use VPR fron toronto university(because is mentioned in log).

Can be an future option to integrate efinix FPGAs in symbiflow toolchain?.

how could I explore/contribute to this option?.

mithro, 2021-02-16:
@eljuligallego - Do you have any contacts at Efinix? I would be happy to talk to them about the topic.

eljuligallego, 2021-02-16:
Hi @mithro,

I have a contact from efinix europe because asked for evaluate the IDE and he was very friendly with good support:

Harald Werner

Efinix

European Sales Director

Breimannweg 3

85391 Allershausen

Germany

haraldw + efinix domain

P.D. I guess you can ask him for the IDE or I can share the files (2GB).

eljuligallego, 2021-09-14:
any news on this?

proppy, 2023-04-14:
met @efxmnalim at a workshop at FOSSASIA, they might be able to do the required connection.