f4pga/f4pga-arch-defs #2058: Trade-off pnr time and performance

andrewb1999, 2021-02-26:
When experimenting with SymbiFlow a long time ago (almost 2 years ago) I was able to adjust vpr parameters such as astar_fac, max_criticality, acc_fac, initial_pres_fac, etc. to trade-off pnr time and performance. After experimenting with this again recently, it seems changing these parameters no longer provides a significant speedup over the default values.

I am wondering if anyone knows a way to decrease pnr time if I am willing to make significant sacrifices to performance. Are there known changes within VPR that have reduced the impact of these input parameters?

litghost, 2021-02-26:
There was an effort to make this trade-off (see https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1735 ).  I'm not surprised you were not able to extract more runtime.

andrewb1999, 2021-02-26:
@litghost I have looked into that work, but my understanding was that it wasn't willing to reduce QoR.  Is that correct?

litghost, 2021-02-26:
> @litghost I have looked into that work, but my understanding was that it wasn't willing to reduce QoR. Is that correct?

That is true.  You can checkout the raw data used in #1735 in https://colab.sandbox.google.com/drive/1X91RGZnvlC7dBjJJUbS7JfqCbPCzJ3Xb?usp=sharing