f4pga/f4pga-arch-defs #2059: Latest architecture links are not uploaded

acomodi, 2021-02-28:
The mechanism is which the latest links of the architectures are generated is the following:

- run all CI's checks on master
- check if all CI's checks are green before
- generate latest links
- push latest links on GCS

It seems that the last stage has some problems due to permissions:

```
Building synchronization state...
Starting synchronization...
Copying file://install/symbiflow-xc7a50t-arty-swbut-overlay_test-latest [Content-Type=application/octet-stream]...
Copying file://install/symbiflow-toolchain-latest [Content-Type=application/octet-stream]...
Copying file://install/symbiflow-xc7a100t_test-latest [Content-Type=application/octet-stream]...
Copying file://install/symbiflow-xc7a50t_test-latest [Content-Type=application/octet-stream]...
/ [0/9 files][    0.0 B/  2.4 KiB]   0% Done                                    
/ [0/9 files][    0.0 B/  2.4 KiB]   0% Done                                    
/ [0/9 files][    0.0 B/  2.4 KiB]   0% Done                                    
/ [0/9 files][    0.0 B/  2.4 KiB]   0% Done                                    
Copying file://install/symbiflow-xc7z010_test-latest [Content-Type=application/octet-stream]...
Copying file://install/symbiflow-benchmarks-latest [Content-Type=application/octet-stream]...
Copying file://install/symbiflow-xc7a50t-arty-switch-processing-overlay_test-latest [Content-Type=application/octet-stream]...
Copying file://install/symbiflow-xc7z020_test-latest [Content-Type=application/octet-stream]...
Copying file://install/symbiflow-xc7a200t_test-latest [Content-Type=application/octet-stream]...
/ [0/9 files][    0.0 B/  2.4 KiB]   0% Done                                    
/ [0/9 files][    0.0 B/  2.4 KiB]   0% Done                                    
/ [0/9 files][    0.0 B/  2.4 KiB]   0% Done                                    
/ [0/9 files][    0.0 B/  2.4 KiB]   0% Done                                    
/ [0/9 files][    0.0 B/  2.4 KiB]   0% Done                                    
AccessDeniedException: 403 ***@foss-fpga-tools-ext.iam.gserviceaccount.com does not have storage.objects.delete access to ***/symbiflow-xc7z010_test-latest.
AccessDeniedException: 403 ***@foss-fpga-tools-ext.iam.gserviceaccount.com does not have storage.objects.delete access to ***/symbiflow-xc7z020_test-latest.
AccessDeniedException: 403 ***@foss-fpga-tools-ext.iam.gserviceaccount.com does not have storage.objects.delete access to ***/symbiflow-xc7a200t_test-latest.
AccessDeniedException: 403 ***@foss-fpga-tools-ext.iam.gserviceaccount.com does not have storage.objects.delete access to ***/symbiflow-toolchain-latest.
AccessDeniedException: 403 ***@foss-fpga-tools-ext.iam.gserviceaccount.com does not have storage.objects.delete access to ***/symbiflow-xc7a50t_test-latest.
AccessDeniedException: 403 ***@foss-fpga-tools-ext.iam.gserviceaccount.com does not have storage.objects.delete access to ***/symbiflow-xc7a50t-arty-swbut-overlay_test-latest.
AccessDeniedException: 403 ***@foss-fpga-tools-ext.iam.gserviceaccount.com does not have storage.objects.delete access to ***/symbiflow-xc7a100t_test-latest.
AccessDeniedException: 403 ***@foss-fpga-tools-ext.iam.gserviceaccount.com does not have storage.objects.delete access to ***/symbiflow-xc7a50t-arty-switch-processing-overlay_test-latest.
AccessDeniedException: 403 ***@foss-fpga-tools-ext.iam.gserviceaccount.com does not have storage.objects.delete access to ***/symbiflow-benchmarks-latest.
CommandException: 9 files/objects
```

Possibly, this is solved by updating the credentials.

mithro, 2021-02-28:
I think I have fixed this now.

litghost, 2021-03-02:
@acomodi Can you see if this is still a problem?

acomodi, 2021-03-02:
We'd need to have a fully green continuous CI to verify this is solved. Unfortunately, the latest merge has a failure in the Vendor tools, preventing the GH actions to proceed and upload the links.

litghost, 2021-03-02:
The nightly should hopefully go green.  Last night failed because of a kokoro failure.

litghost, 2021-03-03:
Nightly last night was green, @acomodi can you double check if the "latest" links updated?

acomodi, 2021-03-03:
Doesn't seem to be working, I see the same error as reported above: https://github.com/SymbiFlow/symbiflow-arch-defs/runs/2023135530?check_suite_focus=true

litghost, 2021-03-03:
@mithro Do you have another idea what the problem might be?

mithro, 2021-03-03:
@QuantamHD is helping me look into this. I think the error message might be a lie...

acomodi, 2021-03-04:
It seems that there was a successful GH action: https://github.com/SymbiFlow/symbiflow-arch-defs/runs/2031341989?check_suite_focus=true

umarcor, 2022-05-18:
I'm closing this because it is related to a deprecated infrastructure (Kokoro), which required a custom syncing solution.

FTR, there is now a different issue with regard to having the latest links/pointers updated: #2655.

closed 2022-05-18 by umarcor