f4pga/f4pga-arch-defs #2108: TinyFPGA BX Bitstreams don't appear to work

donn, 2021-03-17:
I'm trying to get the SOS morse code example for the TinyFPGA BX working. I verified the code as working with icestorm, but with Symbiflow the LED just stays off. I'm running the flow exactly the same as the Makefiles are. I tried a bitstream from my build of symbiflow-arch-defs, but no cigar either.

I'm using these flags with VPR: `--max_router_iterations 500 --routing_failure_predictor off --router_high_fanout_threshold -1 --constant_net_method route --clock_modeling route --route_chan_width 100 --congested_routing_iteration_threshold 0.8 --check_rr_graph off --router_init_wirelength_abort_threshold 2 --allow_unrelated_clustering off --target_ext_pin_util 0.5 --astar_fac 1.0`

---

```v
// look in pins.pcf for all the pin names on the TinyFPGA BX board
module top (
    input CLK,    // 16MHz clock
    output LED,   // User/boot LED next to power LED
    output USBPU  // USB pull-up resistor
);
    // drive USB pull-up resistor to '0' to disable USB
    assign USBPU = 0;

    ////////
    // make a simple blink circuit
    ////////

    // keep track of time and location in blink_pattern
    reg [25:0] blink_counter = 26'b0;

    // pattern that will be flashed over the LED over time
    wire [31:0] blink_pattern = 32'b101010001110111011100010101;

    // increment the blink_counter every clock
    always @(posedge CLK) begin
        blink_counter <= blink_counter + 1;
    end
    
    // light up the LED according to the pattern
    assign LED = blink_pattern[blink_counter[25:21]];
endmodule

```

```pcf
###############################################################################
#
# TinyFPGA BX constraint file (.pcf)
#
###############################################################################
#
# Copyright (c) 2018, Luke Valenty
# All rights reserved.
# 
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions are met:
# 
# 1. Redistributions of source code must retain the above copyright notice, this
#    list of conditions and the following disclaimer.
# 2. Redistributions in binary form must reproduce the above copyright notice,
#    this list of conditions and the following disclaimer in the documentation
#    and/or other materials provided with the distribution.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND
# ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
# WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
# DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT OWNER OR CONTRIBUTORS BE LIABLE FOR
# ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
# (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
# LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND
# ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
# (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
# SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
#
# The views and conclusions contained in the software and documentation are those
# of the authors and should not be interpreted as representing official policies,
# either expressed or implied, of the <project name> project. 
#
###############################################################################

####
# TinyFPGA BX information: https://github.com/tinyfpga/TinyFPGA-BX/
####

# Left side of board
set_io --warn-no-port PIN_1 A2
set_io --warn-no-port PIN_2 A1
set_io --warn-no-port PIN_3 B1
set_io --warn-no-port PIN_4 C2
set_io --warn-no-port PIN_5 C1
set_io --warn-no-port PIN_6 D2
set_io --warn-no-port PIN_7 D1
set_io --warn-no-port PIN_8 E2
set_io --warn-no-port PIN_9 E1
set_io --warn-no-port PIN_10 G2
set_io --warn-no-port PIN_11 H1
set_io --warn-no-port PIN_12 J1
set_io --warn-no-port PIN_13 H2

# Right side of board
set_io --warn-no-port PIN_14 H9
set_io --warn-no-port PIN_15 D9
set_io --warn-no-port PIN_16 D8
set_io --warn-no-port PIN_17 C9
set_io --warn-no-port PIN_18 A9
set_io --warn-no-port PIN_19 B8
set_io --warn-no-port PIN_20 A8
set_io --warn-no-port PIN_21 B7
set_io --warn-no-port PIN_22 A7
set_io --warn-no-port PIN_23 B6
set_io --warn-no-port PIN_24 A6

# SPI flash interface on bottom of board
set_io --warn-no-port SPI_SS F7
set_io --warn-no-port SPI_SCK G7
set_io --warn-no-port SPI_IO0 G6
set_io --warn-no-port SPI_IO1 H7
set_io --warn-no-port SPI_IO2 H4
set_io --warn-no-port SPI_IO3 J8

# General purpose pins on bottom of board
set_io --warn-no-port PIN_25 G1
set_io --warn-no-port PIN_26 J3
set_io --warn-no-port PIN_27 J4
set_io --warn-no-port PIN_28 G9
set_io --warn-no-port PIN_29 J9
set_io --warn-no-port PIN_30 E8
set_io --warn-no-port PIN_31 J2

# LED
set_io --warn-no-port LED B3

# USB
set_io --warn-no-port USBP B4
set_io --warn-no-port USBN A4
set_io --warn-no-port USBPU A3

# 16MHz clock
set_io --warn-no-port CLK B2 # input

```

litghost, 2021-03-17:
The ice40 flow in arch-defs is not actively developed, and I'm not surprised it isn't working.  For ice40, I recommend the nextpnr based flow.

donn, 2021-03-18:
Welp, I'm gonna stop poking at this for now, but for what it's worth, the fasm2asc/asc2fasm utilities are not the culprit. I converted .asc from icestorm to .fasm and back, then packed and programmed it, and it still works.

I'm assuming it's something with VTR at this point, which is out of my depth.