f4pga/f4pga-arch-defs #2124: Kokor CI failing

acomodi, 2021-04-16:
Kokoro CI is failing with the following error message:

```
[00:01:29][ERROR] Config validation failed!
```

It is happening in other repositories as well ([prjxray](https://github.com/SymbiFlow/prjxray) for instance).

[Here](https://source.cloud.google.com/results/invocations/cc1cceb4-245d-4715-ab64-81a1e0bafd2b) is a failing build link

acomodi, 2021-04-16:
cc @mithro @hzeller

acomodi, 2021-04-20:
This was solved. Closing

closed 2021-04-20 by acomodi

seanpmorgan, 2021-05-15:
Hi @acomodi, would you be able to describe what the fix was? Our kokoro build just started failing a day or two ago with the same error and wanted to know if it's anything we can fix or if it's a serverside issue.

acomodi, 2021-05-15:
Hi @seanpmorgan IIRC this was a server side issue. I think that @mithro might have more insights on this one.