f4pga/f4pga-arch-defs #2131: Install configuration for all supported packages

kgugala, 2021-04-24:
Currently we generate package files (like e.g. `pinmap.csv`) only for the devices referenced in our test suite. We should emit those files for all the packages we have in the database.

dnltz, 2021-05-12:
@kgugala - The latest conversation in https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1800 is about the same topic. Unfortunately, litghost is currently not available and I'm not sure how to start.

acomodi, 2021-05-12:
Hi @dnltz, I think that the right place to start is the [add_xc_board](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/a3c75b4b7304a29ff3d48b594e5afe922ce00f7e/xc/common/cmake/device_define.cmake#L1) as discussed in [#1800](https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1800).

At the moment each different part that is available is bound to a board definition, and this is the only way of adding a new part.

There are multiple ways to deal with this issue, and probably one would be to extract the pinmap generation CMake command and detach it from the `board`. This way we can attach the pinmap generation targets in the architecture generation flow and produce all the corresponding pinmaps for each supported part (we can potentially grab the list of all supported parts [from the prjxray-db](https://github.com/SymbiFlow/prjxray-db/blob/master/artix7/mapping/parts.yaml) itself).

The pinmap generation needs to be allowed when specifying an `ROI` type part, when defining a board.

One other note is that, at the moment, there is no directory that is generated to hold all the pinmap files, so it might be necessary to add this functionality as well.

dnltz, 2021-06-27:
@acomodi - Hi, was a little bit busy in the last time and trying to come back to this. So my idea is to generate all pinmaps in the `add_xc_arch_define`. The PROTOTYPE_PATH can be translated to a fabric and afterwards pinmaps for each part in fabric -> device can be generated. This would also require some changes because I would rename the pinmap file from `${BOARD}_pinmap.csv` to `${PART}_pinmap.csv`. Do you think this will work.

> The pinmap generation needs to be allowed when specifying an ROI type part, when defining a board.
I do not understand this totally. Does this mean to keep the pinmap generation for `USE_ROI` and only remove it for `USE_OVERLAY`?

widarr, 2021-11-20:
Hi everyone! I'm new to Symbiflow and have just uploaded my first configuration on an Arty_35 board with the Getting Started guide. The instructions where very detailed and it worked like a treat. I saw a talk on the 35C3 and thought ilI absolutely have to try this before starting with Vivado.

My question for this topic is: I have designed a PCB using an XC7A50T-2FTG256I for a hobby project and I wanted to know how to get or create a pinmap.csv (and or other required files) for this footprint. The arch definitions folder only contains pinmaps for the CSG and CPG versions of the XC7A50T device. Can I get the required files from the Vivado design suite or are they created from a script?

I'm sorry if this is a trivial question but I have just started working my way through the very extensive architecture of Symbiflow :)

dnltz, 2021-11-22:
Hi @widarr - Some pinmaps are currently deployed because they are added manually to https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc/xc7/boards.cmake

This topic and #1800 was about to generate more/all pinmap files. Unfortunately, I'm not super familiar with cmake and had some problems to understand everything totally.

hansfbaier, 2022-01-27:
@acomodi Where can one find the latest downloadable architecture definition file, and how can I build one myself?