f4pga/f4pga-arch-defs #2146: Error yosys-config: line 102: exec: x86_64-conda_cos6-linux-gnu-gcc: not found

the-centry, 2021-05-28:
I want to use the yosys that symbiflow provided to add vhdl plugin but the error happened! What should I do?

the-centry, 2021-05-28:
Can't use the yosys compiled from [yosys](https://github.com/YosysHQ/yosys)?when replace the yosys provided by symbiflow with the compiled one that error happened such as "ERROR: Module $paramod\FDRE\INIT=1'0 contains processes, which are not supported by JSON backend (run `proc` first)"  .

gatecat, 2021-05-28:
The latter problem stems from YosysHQ/yosys#2667, and can be fixed in a similar way to YosysHQ/yosys#2679 by running `blackbox =A:whitebox` before `write_json`.

the-centry, 2021-05-31:
> The latter problem stems from [YosysHQ/yosys#2667](https://github.com/YosysHQ/yosys/issues/2667), and can be fixed in a similar way to [YosysHQ/yosys#2679](https://github.com/YosysHQ/yosys/pull/2679) by running `blackbox =A:whitebox` before `write_json`.

ok! Thanks so much!

closed 2021-06-02 by the-centry