f4pga/f4pga-arch-defs #2147: how to show the graph while placing and routing?

szhou888, 2021-06-09:
If I want to show the graph during placing and routing,how I should do to achieve it?

acomodi, 2021-06-09:
Hi @szhou888, VTR is able to display the place and route process in GUI, but this was not tested in quite a while now in symbiflow.

If you want to give a try, the following should be required:

1. build VTR locally, at the same revision as specified in the [environment file](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/2c411431a41d8afe008bea780628a812c49749e6/environment.yml#L15) (e7d45e013 at the moment of writing). Also, follow this [guide](https://docs.verilogtorouting.org/en/latest/building/building/#dependencies) to install the dependencies to have to enable the GUI.
2. run the following to use the local versions of the tools (which are in the VTR directory):
```
export VPR=<path-to-vpr-binary>
export GENFASM=<path-to-genfasm-binary>
```
3. build a test and add the `--disp on` to the VPR command.

szhou888, 2021-06-10:
> Hi @szhou888, VTR is able to display the place and route process in GUI, but this was not tested in quite a while now in symbiflow.
> 
> If you want to give a try, the following should be required:
> 
> 1. build VTR locally, at the same revision as specified in the [environment file](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/2c411431a41d8afe008bea780628a812c49749e6/environment.yml#L15) (e7d45e013 at the moment of writing). Also, follow this [guide](https://docs.verilogtorouting.org/en/latest/building/building/#dependencies) to install the dependencies to have to enable the GUI.
> 2. run the following to use the local versions of the tools (which are in the VTR directory):
> 
> ```
> export VPR=<path-to-vpr-binary>
> export GENFASM=<path-to-genfasm-binary>
> ```
> 
> 1. build a test and add the `--disp on` to the VPR command.

Follow your guid and failed in show gui with the log file[log](https://github.com/szhou888/symbiflow-arch-defs/blob/master/vpr_gui_log).Could you give some suggestion by the log file?