f4pga/f4pga-arch-defs #2210: XC7 Startupe2 primitive

chriss0612, 2021-07-19:
I wanted to write a controller for the qspi flash on the Nexys 4 DDR / Nexys A7.
To do that i need to instantiate a STARTUPE2 primitive to control the serial clock port that is also used for initial configuration.

I noticed that it wasn't in this repo and tried adding it myself but failed.

I'd like to get some pointer to what is needed for me to be able to add it myself ( or that someone else just adds it themselves)

mithro, 2021-07-19:
FYI - @acomodi @kgugala

tmichalak, 2021-07-29:
@chriss0612 we are working on the documentation on how to add a primitive to arch-defs in this [PR](https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1950). I guess you can use it as a starting point for adding the STARTUPE2 primitive.

chriss0612, 2021-07-29:
Thank you, I will take a look at it!

chriss0612, 2021-07-29:
I've noticed that STARTUPUE2 already is in xc/common/libraries/cells_xtra.xml:

```xml
<module name="STARTUPE2">
    <port name="CFGCLK" type="output" width="1"/>
    <port name="CFGMCLK" type="output" width="1"/>
    <port name="CLK" type="input" width="1"/>
    <port name="EOS" type="output" width="1"/>
    <port name="GSR" type="input" width="1"/>
    <port name="GTS" type="input" width="1"/>
    <port name="KEYCLEARB" type="input" width="1"/>
    <port name="PACK" type="input" width="1"/>
    <port name="PREQ" type="output" width="1"/>
    <port name="USRCCLKO" type="input" width="1"/>
    <port name="USRCCLKTS" type="input" width="1"/>
    <port name="USRDONEO" type="input" width="1"/>
    <port name="USRDONETS" type="input" width="1"/>
    <attribute default="FALSE" name="PROG_USR" type="STRING" values="FALSE, TRUE"/>
    <attribute default="0.0" name="SIM_CCLK_FREQ" type="FLOAT (nS)" values="0.0 to 10.0"/>
  </module>
```
Does that save me any work?

lgl1227, 2024-04-07:
@onestacked  do you add STARTUPUE2 primitive sucessful?

onestacked, 2024-04-07:
I've tried a bit but havn't really gotten far and then stopped working on it.