f4pga/f4pga-arch-defs #2232: BI_DIR for modeling unidirectional wires

stefannikolicns, 2021-07-30:
Hello,

I'm a student of Paolo Ienne at EPFL, doing research on FPGA architecture. We would like to start using the 7-Series models of Symbiflow as a solid reference, but we would also like to incorporate some of their parts into our exploration.

At the moment, our intern Yizhuo and I are trying to get a solid understanding of how the architectures are modeled. What we don't manage to really understand is how the unidirectional wires are represented as bidirectional (also mentioned here, but I couldn't find the answer https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1325#issuecomment-640494265).

In particular, some nodes driving an instance of SS2, shown in ss2.pdf, seem to be only able to drive it at the wrong end (or the SS2 drives them from the wrong end). These I highlighted in red. We also analyzed a bit the routing output of VPR and there are occurrences where a wire indeed seems to be used in the wrong direction (judging by its name). For example, in net.pdf, the NR1 seems to be used as if it were going south. Are you just reusing segment names for wires that could be going in opposite directions?

Both figures are drawn from the data available in the following files: https://drive.switch.ch/index.php/s/6u2jIm9YTBxBL7c

The files in turn were produced by the command in the attached txt file.

Could someone please help us make sense of what is going on here? I'm sure we are missing something, but we can't really figure out what.

Thanks a lot,

Stefan

[ss2.pdf](https://github.com/SymbiFlow/symbiflow-arch-defs/files/6906026/ss2.pdf)
[net.pdf](https://github.com/SymbiFlow/symbiflow-arch-defs/files/6906027/net.pdf)
[vpr_call.txt](https://github.com/SymbiFlow/symbiflow-arch-defs/files/6906028/vpr_call.txt)

stefannikolicns, 2021-07-30:
I wrongly assumed that the north-bound wires go towards the increasing y-coordinate, whereas it is clear from [utils/tile_splitter/grid.py](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/ca89898ab44cb5fd429968cc99a60feb53cb22c9/utils/tile_splitter/grid.py#L62) that this is not the case. Flipping the vertical wires in the attached figures puts everything in order.

closed 2021-07-30 by stefannikolicns

mithro, 2021-07-30:
@stefannikolicns - If there is documentation that should be updated to make this clearer, we would love a patch!