f4pga/f4pga-arch-defs #2237: Support for programming tool - openFPGALoader

lpawelcz, 2021-08-02:
[OpenFPGALoader](https://github.com/trabucayre/openFPGALoader) is an universal FPGA programming tool which supports a variety of FPGA devices and programmers based on JTAG, DAP interface, ORBTrace, DFU and FTDI chips.
As OpenFPGALoader is now under an Apache 2.0 license, we should consider supporting this tool. It might even make sense for OpenFPGALoader to be the default programming tool instead of OpenOCD.

umarcor, 2021-08-07:
> It might even make sense for OpenFPGALoader to be the default programming tool instead of OpenOCD.

I think that openFPGALoader cannot be considered a replacement for OpenOCD (yet). openFPGALoader is a memory programming utility, while OpenOCD is an On-Chip Debugger. Therefore, users dealing with SoCs will still need OpenOCD for interacting with the CPU. See https://github.com/trabucayre/openFPGALoader/pull/65#issuecomment-881898746.

mithro, 2021-08-07:
I agree that we will need to support both OpenOCD and OpenFPGALoader.