f4pga/f4pga-arch-defs #2317: Wrong synthesis result when buses are not 0-addressed

gergoerdi, 2021-11-12:
This is in relation to https://github.com/YosysHQ/yosys/issues/3078 where I was told that the debug output of Yosys is just misleading, i.e. the problem should be elsewhere in the toolchain.

Given the following Verilog input:

```
module Top(
           output [5:2] LED
           );

   assign LED[5:2] = 4'b1101;
endmodule
```

I am synthesizing for the Nexys A7 50T dev board with a Xilinx xc7a50tcsg324 FPGA. I connect `LED[5:2]` to the LEDs on board with the following XDC:

```
set_property -dict { PACKAGE_PIN J13   IOSTANDARD LVCMOS33 } [get_ports { LED[2] }];
set_property -dict { PACKAGE_PIN N14   IOSTANDARD LVCMOS33 } [get_ports { LED[3] }];
set_property -dict { PACKAGE_PIN R18   IOSTANDARD LVCMOS33 } [get_ports { LED[4] }];
set_property -dict { PACKAGE_PIN V17   IOSTANDARD LVCMOS33 } [get_ports { LED[5] }];
```

If I synthesize this with Vivado, LEDs 2, 4 and 5 light up, as expected. If I synthesize this with Symbiflow, instead only LED 2 lights up. 

Full reproduer is at https://github.com/gergoerdi/yosys-issue-3078/tree/1eb7b9a88b678087e7ace0e3ade7c45e13e39c34

mithro, 2021-11-12:
```eblif
# Generated by Yosys 0.9+4270 (git sha1 539d4ee9, x86_64-conda_cos6-linux-gnu-gcc 1.24.0.133_b0863d8_dirty -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -fdebug-prefix-map=/home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/yosys_1628927583200/work=/usr/local/src/conda/yosys-0.9_5567_g539d4ee9 -fdebug-prefix-map=/home/cactus/.conda/envs/xc7=/usr/local/src/conda-prefix -fPIC -Os -fno-merge-constants)

.model Top
.inputs
.outputs LED[2] LED[3] LED[4] LED[5]
.subckt GND GND=$false
.subckt VCC VCC=$true
.subckt VCC VCC=$undef
.subckt T_INV TI=$true TO=$iopadmap$Top.LED.t
.cname $iopadmap$Top.LED.genblk1.t_inv
.attr module_not_derived 00000000000000000000000000000001
.attr src "/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:1921.5-1925.4|/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:1950.11-1953.6"
.subckt OBUFT_VPR I=$true O=LED[2] T=$iopadmap$Top.LED.t
.cname $iopadmap$Top.LED.obuft
.attr module_not_derived 00000000000000000000000000000001
.attr src "/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:1921.5-1925.4|/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:2064.5-2068.4"
.param DRIVE 00000000000000000000000000001100
.param IOSTANDARD "LVCMOS33"
.param IO_LOC_PAIRS "NONE"
.param LVCMOS12_DRIVE_I12 0
.param LVCMOS12_DRIVE_I4 0
.param LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SLEW_FAST 0
.param LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135_SSTL15_SLEW_SLOW 1
.param LVCMOS12_LVCMOS15_LVCMOS18_SSTL135_SSTL15_STEPDOWN 0
.param LVCMOS12_LVCMOS25_DRIVE_I8 0
.param LVCMOS15_DRIVE_I12 0
.param LVCMOS15_DRIVE_I8 0
.param LVCMOS15_LVCMOS18_LVCMOS25_DRIVE_I4 0
.param LVCMOS15_SSTL15_DRIVE_I16_I_FIXED 0
.param LVCMOS18_DRIVE_I12_I8 0
.param LVCMOS18_DRIVE_I16 0
.param LVCMOS18_DRIVE_I24 0
.param LVCMOS25_DRIVE_I12 0
.param LVCMOS25_DRIVE_I16 0
.param LVCMOS33_DRIVE_I16 0
.param LVCMOS33_LVTTL_DRIVE_I12_I16 1
.param LVCMOS33_LVTTL_DRIVE_I12_I8 0
.param LVCMOS33_LVTTL_DRIVE_I4 0
.param LVTTL_DRIVE_I24 0
.param PULLTYPE "NONE"
.param PULLTYPE_KEEPER 0
.param PULLTYPE_NONE 1
.param PULLTYPE_PULLDOWN 0
.param PULLTYPE_PULLUP 0
.param SLEW "SLOW"
.param SSTL135_DRIVE_I_FIXED 0
.param SSTL135_SSTL15_SLEW_FAST 0
.subckt T_INV TI=$true TO=$iopadmap$Top.LED_1.t
.cname $iopadmap$Top.LED_1.genblk1.t_inv
.attr module_not_derived 00000000000000000000000000000001
.attr src "/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:1921.5-1925.4|/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:1950.11-1953.6"
.subckt OBUFT_VPR I=$false O=LED[3] T=$iopadmap$Top.LED_1.t
.cname $iopadmap$Top.LED_1.obuft
.attr module_not_derived 00000000000000000000000000000001
.attr src "/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:1921.5-1925.4|/home/cactus/prog/fpga/symbiflow/symbiflow-examples/xc7/install/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:2064.5-2068.4"
.param DRIVE 00000000000000000000000000001100
.param IOSTANDARD "LVCMOS33"
.param IO_LOC_PAIRS "NONE"
.param LVCMOS12_DRIVE_I12 0
.param LVCMOS12_DRIVE_I4 0
.param LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SLEW_FAST 0
.param LVCMOS12_LVCMOS15_LVCMOS18_LVCMOS25_LVCMOS33_LVTTL_SSTL135_SSTL15_SLEW_SLOW 1
.param LVCMOS12_LVCMOS15_LVCMOS18_SSTL135_SSTL15_STEPDOWN 0
.param LVCMOS12_LVCMOS25_DRIVE_I8 0
.param LVCMOS15_DRIVE_I12 0
.param LVCMOS15_DRIVE_I8 0
.param LVCMOS15_LVCMOS18_LVCMOS25_DRIVE_I4 0
.param LVCMOS15_SSTL15_DRIVE_I16_I_FIXED 0
.param LVCMOS18_DRIVE_I12_I8 0
[... 72 lines cut ...]
.param LVCMOS12_LVCMOS25_DRIVE_I8 0
.param LVCMOS15_DRIVE_I12 0
.param LVCMOS15_DRIVE_I8 0
.param LVCMOS15_LVCMOS18_LVCMOS25_DRIVE_I4 0
.param LVCMOS15_SSTL15_DRIVE_I16_I_FIXED 0
.param LVCMOS18_DRIVE_I12_I8 0
.param LVCMOS18_DRIVE_I16 0
.param LVCMOS18_DRIVE_I24 0
.param LVCMOS25_DRIVE_I12 0
.param LVCMOS25_DRIVE_I16 0
.param LVCMOS33_DRIVE_I16 0
.param LVCMOS33_LVTTL_DRIVE_I12_I16 1
.param LVCMOS33_LVTTL_DRIVE_I12_I8 0
.param LVCMOS33_LVTTL_DRIVE_I4 0
.param LVTTL_DRIVE_I24 0
.param PULLTYPE "NONE"
.param PULLTYPE_KEEPER 0
.param PULLTYPE_NONE 1
.param PULLTYPE_PULLDOWN 0
.param PULLTYPE_PULLUP 0
.param SLEW "SLOW"
.param SSTL135_DRIVE_I_FIXED 0
.param SSTL135_SSTL15_SLEW_FAST 0
.end
```

mithro, 2021-11-12:
> ```
> # Generated by Yosys 0.9+4270 (git sha1 539d4ee9, x86_64-conda_cos6-linux-gnu-gcc 1.24.0.133_b0863d8_dirty -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -fdebug-prefix-map=/home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/yosys_1628927583200/work=/usr/local/src/conda/yosys-0.9_5567_g539d4ee9 -fdebug-prefix-map=/home/cactus/.conda/envs/xc7=/usr/local/src/conda-prefix -fPIC -Os -fno-merge-constants)
> 
> .model Top
> .inputs
> .outputs LED[2] LED[3] LED[4] LED[5]
> .subckt GND GND=$false
> .subckt VCC VCC=$true
> .subckt VCC VCC=$undef
>
> .subckt T_INV TI=$true TO=$iopadmap$Top.LED.t
> .cname $iopadmap$Top.LED.genblk1.t_inv
>
> .subckt OBUFT_VPR I=$true O=LED[2] T=$iopadmap$Top.LED.t
> .cname $iopadmap$Top.LED.obuft
> .param IO_LOC_PAIRS "NONE"
>
> .subckt T_INV TI=$true TO=$iopadmap$Top.LED_1.t
> .cname $iopadmap$Top.LED_1.genblk1.t_inv
>
> .subckt OBUFT_VPR I=$false O=LED[3] T=$iopadmap$Top.LED_1.t
> .cname $iopadmap$Top.LED_1.obuft
> .param IO_LOC_PAIRS "NONE"
>
> .subckt T_INV TI=$true TO=$iopadmap$Top.LED_2.t
> .cname $iopadmap$Top.LED_2.genblk1.t_inv
>
> .subckt OBUFT_VPR I=$true O=LED[4] T=$iopadmap$Top.LED_2.t
> .cname $iopadmap$Top.LED_2.obuft
> .param IO_LOC_PAIRS "LED[2]:J13"
>
> .subckt OBUFT_VPR I=$true O=LED[5] T=$iopadmap$Top.LED_3.t
> .cname $iopadmap$Top.LED_3.obuft
> .param IO_LOC_PAIRS "LED[3]:N14"
> .end
> ```