f4pga/f4pga-arch-defs #2319: Conda lock pull requests not running CI?

mithro, 2021-11-16:
See for example https://github.com/SymbiFlow/symbiflow-arch-defs/pull/2318

Seems to be missing kokoro and GitHub Actions run?

mithro, 2021-11-16:
@ajelinski and @acomodi

kgugala, 2021-11-16:
I suppose the kokoro run is not triggered automatically because the author of the PR (bot) is not a member of the project - maybe we should automatically mark the PRs with `kokoro:force-run` label?

Missing GH-A CI is weird we have to check this

kgugala, 2021-11-16:
https://github.com/peter-evans/create-pull-request/issues/48 <-- this one seems to explain the behaviour we see and it proposes workaround

mithro, 2021-11-16:
Can we use a deploy key to push as a Robot Account?

acomodi, 2021-11-17:
@mithro Yeah, I think that we need to:

- [Generate a PAT](https://github.com/settings/tokens) for https://github.com/symbiflow-robot
- Add the generated PAT as a secret to symbiflow-arch-defs
- Add it to the [conda lock action variables](https://github.com/SymbiFlow/actions/blob/3fdc436e761a6312c3e0a554f18b727c34434106/update_conda_lock/create_pull_request.py#L62)

ajelinski, 2021-11-17:
You're both correct @acomodi and @kgugala, the GH Actions token used currently is by design unable to trigger the CI.

mithro, 2021-11-18:
Couple of links from a discussion with @acomodi;
 - https://github.com/puppetlabs/peter-evans-create-pull-request/blob/master/docs/concepts-guidelines.md#push-using-ssh-deploy-keys
 - https://github.com/mithro/openroad-actions/blob/main/push_to/action.yml
 - https://github.com/mithro/openroad-actions/commit/658b9c8199f7990b1cb36c5da01182e20c1dbdb8#diff-20e3a22481d805ead30527a5ee1e4669918f1e692cc321265ae11377c6936074