f4pga/f4pga-arch-defs #2324: Xilinx vendor tests are failing on master?

mithro, 2021-11-19:
https://github.com/SymbiFlow/symbiflow-arch-defs/commit/ef6fff3c0ac33035fc67a61eac782273d918adbe is failing "Xilinx Series 7 - Vendor Tool Tests — Build failed" see https://source.cloud.google.com/results/invocations/65e21bb7-a598-4bc2-94d8-a089d132d562/log

mithro, 2021-11-19:
```
FAILED: cd /tmpfs/src/github/symbiflow-arch-defs-continuous-xc7-vendor/build/xc/xc7/tests/ddr/ddr_uart_arty100t/artix7_100t-xc7a100t-virt-xc7a100t-test && /tmpfs/src/github/symbiflow-arch-defs-continuous-xc7-vendor/env/conda/envs/symbiflow_arch_def_base/bin/cmake -E remove -f ddr_uart_arty100t/artix7_100t-xc7a100t-virt-xc7a100t-test/design_ddr_uart_arty100t_vivado.dcp && /tmpfs/src/github/symbiflow-arch-defs-continuous-xc7-vendor/env/conda/envs/symbiflow_arch_def_base/bin/cmake -E remove -f ddr_uart_arty100t/artix7_100t-xc7a100t-virt-xc7a100t-test/design_ddr_uart_arty100t_vivado.xpr && /tmpfs/src/github/symbiflow-arch-defs-continuous-xc7-vendor/third_party/prjxray/utils/vivado.sh -mode batch -source /tmpfs/src/github/symbiflow-arch-defs-continuous-xc7-vendor/build/xc/xc7/tests/ddr/ddr_uart_arty100t_vivado_runme.tcl > /tmpfs/src/github/symbiflow-arch-defs-continuous-xc7-vendor/build/xc/xc7/tests/ddr/ddr_uart_arty100t/artix7_100t-xc7a100t-virt-xc7a100t-test/vivado.stdout.log
ERROR: [DRC RTRES-2] Global clock nets are using local routing resources: Global clock nets are using local routing resources. 1 net(s) have at least one node with COST_CODE_NAME equal to BOUNCEACROSS or INTENT_CODE_NAME equal to NODE_DOUBLE, NODE_HLONG, NODE_HQUAD, NODE_SINGLE, NODE_VLONG, or NODE_VQUAD. This situation occurs when a global clock net must enter a clock region where all global clock routing resources are occupied. Check the report from report_clock_utilization to determine which clock regions are traversed by the problem net and use floorplanning or other physical constraints to ensure a maximum of 12 global clock nets occupy each clock region. The problem net(s) are clk200_clk.
ERROR: [Vivado 12-1345] Error(s) found during DRC. Bitgen not run.
```

umarcor, 2022-04-06:
@mithro @acomodi I guess this was fixed already, so it can be closed?