f4pga/f4pga-arch-defs #2331: symbiflow_route needs to fail when timing is not met

tcal-x, 2021-11-24:
The `symbiflow_route` wrapper needs to determine if VPR met the desired timing, and exit with an error status if it does not.   Perhaps we can add an option for skipping the timing check, but the default should be that timing is checked, and the flow fails if timing is not met.

See https://github.com/enjoy-digital/litex/issues/1108.

mithro, 2021-11-24:
@kgugala / @acomodi -- PTAL.

acomodi, 2021-11-25:
I believe that, rather than having this check in the wrapper, this should be within VPR, and the option passed to it to hard fail in case of timing violations produced during the analysis step.

mithro, 2021-11-25:
As I have said multiple times, timing failure **must** cause the tooling to exit with an error by default. There can be an option to allow it to continue with a timing fail.

acomodi, 2021-11-26:
My suggestion was to keep this check within VPR, rather than in the wrapper. Not sure if VTR devs will want to fail by default in case of timing violations during the analysis step, but in case we might just add the option in the wrapper. Either way this should end up in VPR

umarcor, 2022-05-18:
I'm closing this because the wrappers were removed from this repo, and the issue is tracked in verilog-to-routing/vtr-verilog-to-routing#1928.

tcal-x, 2022-08-15:
Followup: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2060 has been merged, so we can now pick up work on correct handling in the F4PGA flow.