f4pga/f4pga-arch-defs #2344: Process of adding a new architecture

apond308, 2021-12-09:
I am looking to add a new architecture, and cannot find much documentation on how the process works. I looked around in the testarch directory, but after running build for the [4x4 device](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/testarch/devices/CMakeLists.txt) (clutff-unidir-s4), I can't find any results or outputs from the build.

mithro, 2022-01-04:
I'm unsure what the question here is?

apond308, 2022-02-03:
I am looking to add a new architecture (SOFA) to SymbiFlow, and can't find any documentation on how to implement a new architecture on symbiflow-arch-defs.

I copied the structure of the 'testarch' directory, and replaced those VPR arch files with mine, and it compiles fine when I add this new architecture to the main CMakeLists.txt file. I do not know, however, how to implement tests or run any benchmarks on the new architecture I just put in. The 'testarch' directory doesn't have any tests either

I am looking for some direction on how to implement new tests for an architecture in symbiflow-arch-defs.

KKtiandao, 2022-04-14:
> I am looking for some direction on how to implement tests for an architecture in symbiflow-arch-defs.
@apond308 hi, I have a same issue after added a new architecture (spartan7). Do you find any method to verify the architecture newly added?

KKtiandao, 2022-04-14:
@apond308 I also had considered directory 'testarch' to be for verification. But after some research, I think it maybe used to explore new archs.