f4pga/f4pga-arch-defs #2348: Error instantiating qlal4s3b_cell_macro with new names (Clk_Cxx)

carlosedp, 2021-12-15:
I'm instantiating the macro for clock and it fails packing. This is my Verilog:

```verilog
qlal4s3b_cell_macro u_qlal4s3b_cell_macro (
    .Clk_C16 (clko),
    .Clk_C16_Rst (),
    .Clk_C21     (),
    .Clk_C21_Rst ()
);
```

The error I'm getting is:

```
Type: Blif file
File: Toplevel.eblif
Line: 1989
Message: Failed to find matching architecture model for 'qlal4s3b_cell_macro'
```

Versions:

```
root@eb74749d0141:/src# yosys --version
Yosys 0.9+2406 (git sha1 d282be04, x86_64-conda_cos6-linux-gnu-gcc 1.24.0.133_b0863d8_dirty -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -fdebug-prefix-map=/home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/quicklogic-yosys_1625138793175/work=/usr/local/src/conda/quicklogic-yosys-0.8.0_105_gd282be04 -fdebug-prefix-map=/usr/local/envs/eos-s3=/usr/local/src/conda-prefix -fPIC -Os)
root@eb74749d0141:/src#
root@eb74749d0141:/src# vpr --version
VPR FPGA Placement and Routing.
Version: 8.1.0-dev+e73e88940
Revision: 8.0.0-4023-ge73e88940
Compiled: 2021-07-01T11:51:56
Compiler: GNU 9.3.0 on Linux-4.15.0-1113-azure x86_64
Build Info: Release IPO PGO VTR_ASSERT_LEVEL=2
```

carlosedp, 2022-01-11:
Ping... any news about this?
Might be related to https://symbiflow.slack.com/archives/CGCL3DBMM/p1639411001290300

Aminekachkach, 2023-04-12:
Hello,
could you resolve the issue?
Also the link you suggest requires an account is it normal?
Br

Aminekachkach, 2023-04-12:
> Ping... any news about this? Might be related to https://symbiflow.slack.com/archives/CGCL3DBMM/p1639411001290300
> 
> > **[Slack](https://symbiflow.slack.com/archives/CGCL3DBMM/p1639411001290300)**

Hello,
could you resolve the issue?
Also the link you suggest requires an account is it normal?
Br