f4pga/f4pga-arch-defs #2351: arch-defs not downloading the latest version of prjxray-db when building

MidsummerNight, 2021-12-20:
Right now after running `make env` in `symbiflow-arch-defs` the `prjxray-db` downloaded is quite outdated (notably missing the new [spartan7](https://github.com/SymbiFlow/prjxray-db/tree/master/spartan7) info, which I intend to use for adding Spartan-7 support to `arch-defs`).

How can I let `arch-defs` always pull the latest `prjxray-db` data upon building? I've been trying to track down why `make env` keeps downloading the old version but so far no luck. @acomodi @mithro @mkurc-ant @kgugala Any ideas? Thanks!

acomodi, 2021-12-20:
Hi @MidsummerNight, in theory the package version [specified in the conda-lock](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/conda_lock.yml#L65) is pointing to the version that includes the spartan7 changes. Maybe a clean build might solve the issue?

MidsummerNight, 2021-12-20:
@acomodi Yeah that's what's bothering me. I did clone a clean new copy of `arch-defs` from Github to a brand new directory, then ran `make env` in that new directory, but no, the new `prjxray-db` with spartan7 was not downloaded. Can confirm not even a single `spartan7` folder exists:

```
me@me-machine:~/arch-defs$ find . -type d -name spartan7
me@me-machine:~/arch-defs$ find . -type d -name artix7
./symbiflow-arch-defs/third_party/symbiflow-xc-fasm2bels/third_party/prjxray-db/artix7
./symbiflow-arch-defs/third_party/symbiflow-xc-fasm2bels/third_party/prjxray/database/artix7
./symbiflow-arch-defs/third_party/symbiflow-xc-fasm2bels/third_party/prjxray/settings/artix7
./symbiflow-arch-defs/third_party/symbiflow-xc-fasm2bels/third_party/prjxray/fuzzers/074-dump_all/ignored_wires/artix7
./symbiflow-arch-defs/third_party/symbiflow-xc-fasm2bels/third_party/prjxray/third_party/display_port/src/artix7
./symbiflow-arch-defs/third_party/prjxray/database/artix7
./symbiflow-arch-defs/third_party/prjxray/settings/artix7
./symbiflow-arch-defs/third_party/prjxray/fuzzers/074-dump_all/ignored_wires/artix7
./symbiflow-arch-defs/third_party/prjxray/third_party/display_port/src/artix7
./symbiflow-arch-defs/xc/xc7/archs/artix7
./symbiflow-arch-defs/build/xc/xc7/archs/artix7
./symbiflow-arch-defs/env/downloads/conda-pkgs/prjxray-db-0.0_257_g0a0adde-20211214_154543/share/symbiflow/prjxray-db/artix7
./symbiflow-arch-defs/env/conda/envs/symbiflow_arch_def_base/share/symbiflow/prjxray-db/artix7
me@me-machine:~/arch-defs$ ls -alk
total 12
drwxrwxr-x  3 me me 4096 Dec 20 16:32 .
drwxr-xr-x 51 me me 4096 Dec 20 17:15 ..
drwxrwxr-x 18 me me 4096 Dec 20 16:59 symbiflow-arch-defs
me@me-machine:~/arch-defs$ 
```

acomodi, 2021-12-20:
@MidsummerNight Right, discovered the issue and opened a PR in the conda build script: https://github.com/hdl/conda-eda/pull/151

MidsummerNight, 2021-12-20:
@acomodi Okay thanks! Once your PR gets merged into `conda-eda` I should be able to get the `prjxray-db` with spartan7 info by  doing a clean download and build for `arch-defs` right?

acomodi, 2021-12-20:
Might take a bit more, as the flow is the following:

- A scheduled GH action builds the conda packages once every day, and uploads them
- Another scheduled GH action in arch-defs checks for new conda-packages and opens a PR to update the packages to the latest (e.g. https://github.com/SymbiFlow/symbiflow-arch-defs/pull/2345)
- When the BOT PR gets merged, the package will be available.

You can also check probably tomorrow the latest package from conda https://anaconda.org/LiteX-Hub/prjxray-db/files, if it is there, you can just grab it and update the `conda_lock.yml` file.

MidsummerNight, 2022-01-25:
Conda now downloads the latest `prjxray-db` with `spartan7`. I think this issue can be closed now.

closed 2022-01-25 by MidsummerNight