f4pga/f4pga-arch-defs #2363: Check-status CI job is failing?

mithro, 2022-01-12:
https://github.com/SymbiFlow/symbiflow-arch-defs/actions/runs/1689091015

```
Run ./.github/check-status
Building docker image
  Dockerfile for action: '/home/runner/work/symbiflow-arch-defs/symbiflow-arch-defs/./.github/check-status/Dockerfile'.
  /usr/bin/docker build -t 9916a7:d2bdedf2b8c554ce2aed8802e3176a61 -f "/home/runner/work/symbiflow-arch-defs/symbiflow-arch-defs/./.github/check-status/Dockerfile" "/home/runner/work/symbiflow-arch-defs/symbiflow-arch-defs/.github/check-status"
  Sending build context to Docker daemon  6.144kB
  
  Step 1/5 : FROM python:alpine
  alpine: Pulling from library/python
  59bf1c3509f3: Pulling fs layer
  07a400e93df3: Pulling fs layer
  d8c681bc2183: Pulling fs layer
  c7f1ca1171d9: Pulling fs layer
  dc9256fcea2e: Pulling fs layer
  c7f1ca1171d9: Waiting
  dc9256fcea2e: Waiting
  07a400e93df3: Verifying Checksum
  07a400e93df3: Download complete
  59bf1c3509f3: Verifying Checksum
  59bf1c3509f3: Download complete
  d8c681bc2183: Verifying Checksum
  d8c681bc2183: Download complete
  59bf1c3509f3: Pull complete
  c7f1ca1171d9: Verifying Checksum
  c7f1ca1171d9: Download complete
  dc9256fcea2e: Verifying Checksum
  dc9256fcea2e: Download complete
  07a400e93df3: Pull complete
  d8c681bc2183: Pull complete
  c7f1ca1171d9: Pull complete
  dc9256fcea2e: Pull complete
  Digest: sha256:dce56d40d885d2c8847aa2a278a29d50450c8e3d10f9d7ffeb2f38dcc1eb0ea4
  Status: Downloaded newer image for python:alpine
   ---> eb5bc7d10d52
  Step 2/5 : RUN apk add git && pip install PyGithub git+https://github.com/SymbiFlow/symbiflow-tools-data-manager#egg=stdm --progress-bar off
   ---> Running in 8367a438dd8b
  fetch https://dl-cdn.alpinelinux.org/alpine/v3.15/main/x86_64/APKINDEX.tar.gz
  fetch https://dl-cdn.alpinelinux.org/alpine/v3.15/community/x86_64/APKINDEX.tar.gz
  (1/5) Installing brotli-libs (1.0.9-r5)
  (2/5) Installing nghttp2-libs (1.46.0-r0)
  (3/5) Installing libcurl (7.80.0-r0)
  (4/5) Installing pcre2 (10.39-r0)
  (5/5) Installing git (2.34.1-r0)
  Executing busybox-1.34.1-r3.trigger
  OK: 27 MiB in 41 packages
  Collecting stdm
    Cloning https://github.com/SymbiFlow/symbiflow-tools-data-manager to /tmp/pip-install-h2a49umk/stdm_71c6c1d45eff47ceb2032d889a0c6412
    Running command git clone -q https://github.com/SymbiFlow/symbiflow-tools-data-manager /tmp/pip-install-h2a49umk/stdm_71c6c1d45eff47ceb2032d889a0c6412
    Resolved https://github.com/SymbiFlow/symbiflow-tools-data-manager to commit faafe2e0128cd6a4f174f69fc1e31d3bc385be09
  Collecting PyGithub
    Downloading PyGithub-1.55-py3-none-any.whl (291 kB)
  Collecting requests>=2.14.0
    Downloading requests-2.27.1-py2.py3-none-any.whl (63 kB)
  Collecting deprecated
    Downloading Deprecated-1.2.13-py2.py3-none-any.whl (9.6 kB)
  Collecting pyjwt>=2.0
    Downloading PyJWT-2.3.0-py3-none-any.whl (16 kB)
  Collecting pynacl>=1.4.0
    Downloading PyNaCl-1.5.0-cp36-abi3-musllinux_1_1_x86_64.whl (1.1 MB)
  Collecting cffi>=1.4.1
    Downloading cffi-1.15.0.tar.gz (484 kB)
  Collecting pycparser
    Downloading pycparser-2.21-py2.py3-none-any.whl (118 kB)
  Collecting certifi>=2017.4.17
    Downloading certifi-2021.10.8-py2.py3-none-any.whl (149 kB)
  Collecting idna<4,>=2.5
    Downloading idna-3.3-py3-none-any.whl (61 kB)
  Collecting charset-normalizer~=2.0.0
[... 101 lines cut ...]
      copying cffi/cffi_opcode.py -> build/lib.linux-x86_64-3.10/cffi
      copying cffi/verifier.py -> build/lib.linux-x86_64-3.10/cffi
      copying cffi/cparser.py -> build/lib.linux-x86_64-3.10/cffi
      copying cffi/vengine_cpy.py -> build/lib.linux-x86_64-3.10/cffi
      copying cffi/recompiler.py -> build/lib.linux-x86_64-3.10/cffi
      copying cffi/setuptools_ext.py -> build/lib.linux-x86_64-3.10/cffi
      copying cffi/commontypes.py -> build/lib.linux-x86_64-3.10/cffi
      copying cffi/_cffi_include.h -> build/lib.linux-x86_64-3.10/cffi
      copying cffi/parse_c_type.h -> build/lib.linux-x86_64-3.10/cffi
      copying cffi/_embedding.h -> build/lib.linux-x86_64-3.10/cffi
      copying cffi/_cffi_errors.h -> build/lib.linux-x86_64-3.10/cffi
      running build_ext
      building '_cffi_backend' extension
      creating build/temp.linux-x86_64-3.10
      creating build/temp.linux-x86_64-3.10/c
      gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -DTHREAD_STACK_SIZE=0x100000 -fPIC -I/usr/include/ffi -I/usr/include/libffi -I/usr/local/include/python3.10 -c c/_cffi_backend.c -o build/temp.linux-x86_64-3.10/c/_cffi_backend.o
      error: command 'gcc' failed: No such file or directory
      ----------------------------------------
  ERROR: Command errored out with exit status 1: /usr/local/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-h2a49umk/cffi_03b56e3c4dd7414db689dc35aed00e51/setup.py'"'"'; __file__='"'"'/tmp/pip-install-h2a49umk/cffi_03b56e3c4dd7414db689dc35aed00e51/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-714lnhio/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.10/cffi Check the logs for full command output.
  WARNING: You are using pip version 21.2.4; however, version 21.3.1 is available.
  You should consider upgrading via the '/usr/local/bin/python -m pip install --upgrade pip' command.
  The command '/bin/sh -c apk add git && pip install PyGithub git+https://github.com/SymbiFlow/symbiflow-tools-data-manager#egg=stdm --progress-bar off' returned a non-zero code: 1
  
Error: Docker build failed with exit code 1
```

mithro, 2022-01-12:
Issue seems to be missing GCC? - `    error: command 'gcc' failed: No such file or directory`

mithro, 2022-01-12:
Last working run -> https://github.com/SymbiFlow/symbiflow-arch-defs/actions/runs/1668696179

mithro, 2022-01-12:
Looks like missing gcc was always failing when previously succeeding - no idea why it started failing the CI.

mithro, 2022-01-12:
@acomodi - Any idea?

acomodi, 2022-01-13:
@mithro not sure why it started to fail now, but installing gcc and some other required libs seem to solve the problem: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/2365

closed 2022-01-14 by acomodi