f4pga/f4pga-arch-defs #2368: xc7a50t_test database does not contain the part xc7a35tftg256-1

hansfbaier, 2022-01-26:
It looks like the mentioned part is already supported by prjxray:
prjxray/settings/resources.yaml contains it:
```yaml
xc7a35tfgg484-2L: *id032
xc7a35tfgg484-3: *id032
xc7a35tftg256-1:
  pins: &id033
    0: N13
    1: K12
    2: F15
    3: K5
```
but the downloadable database for symbiflow does not contain this footprint variant.
What could I do to add this part to the database?

acomodi, 2022-01-26:
@mithro @hansfbaier At the moment the generation of the package files, thus adding support for another part belonging to the same device (e.g. xc7a35t) is still a manual process.

There is an [issue open](https://github.com/SymbiFlow/symbiflow-arch-defs/issues/2131)  to make this automatic, so to grab data from x-ray and populate all the parts' data without having to do it manually, but it hasn't been tackled yet.

hansfbaier, 2022-01-26:
This is a duplicate of #1800

hansfbaier, 2022-01-26:
@acomodi Thanks for the reference!

closed 2022-01-26 by hansfbaier