f4pga/f4pga-arch-defs #2372: Need help adding support for Spartan-7 board to arch-defs

MidsummerNight, 2022-01-29:
Hi there,

Since the `prjxray-db` of Spartan-7 was generated last month, I've been trying to add support for Spartan-7 to `arch-defs`. Following @acomodi's [instructions](https://github.com/SymbiFlow/prjxray/issues/1789#issuecomment-994546002), I add directories and files for Spartan-7, along with a simple blinking LED test targeting the Alinx [AX7050](https://www.amazon.com/ALINX-Spartan-7-Development-Platform-XC7S50FGGA484I/dp/B07DBMX8FR) board (which carries the `xc7s50fgga484-1` part). Both the [HDL source code](https://github.com/MidsummerNight/symbiflow-arch-defs/blob/673568f829d79d4971698f4d99cc08eacf3e7830/xc/xc7/tests/counter/counter_ax7050.v) and [XDC constraint file](https://github.com/MidsummerNight/symbiflow-arch-defs/blob/673568f829d79d4971698f4d99cc08eacf3e7830/xc/xc7/tests/common/counter_ax7050.xdc) have been successfully implemented in Vivado and runs correctly on the board, so no problems there.

I successfully ran `make build` and found `build/xc/xc7/tests/counter/Makefile` indeed has the `counter_ax7050_bit` target. However, when I ran `make counter_ax7050_bit`, I got an error saying `No rule to make target xc/xc7/tests/common/counter_ax7050.xdc` that was needed to create `top_synth.json`. Upon further inspection, I found that the recipe for `counter_ax7050_bit.dir/all` in `Makefile2` somehow does not include `counter_ax7050.xdc`. I wrote the `CMakeLists.txt` for `tests/counter` like [this](https://github.com/MidsummerNight/symbiflow-arch-defs/blob/7bc40406d3c6a80b85ba2c72709b0fa09cd6bff2/xc/xc7/tests/counter/CMakeLists.txt). Did I write it correctly? Or did something else cause my XDC file to not be in the recipe?

Thanks!

MidsummerNight, 2022-01-29:
What is [INPUT_XDC_FILES](https://github.com/MidsummerNight/symbiflow-arch-defs/blob/7bc40406d3c6a80b85ba2c72709b0fa09cd6bff2/xc/xc7/tests/counter/CMakeLists.txt#L188) for? Should I use `INPUT_IO_FILE` instead? And why does INPUT_XDC_FILES always comes with INPUT_IO_FILE in the existing targets?

mithro, 2022-01-29:
@acomodi / @kgugala - Can you help here?

acomodi, 2022-01-31:
Hi @MidsummerNight, thanks for the effort in adding support for the Spartan7 devices!

I suggest opening a PR with your WIP work, as it can run through CI, and is generally easier to understand the diffs.

Looking at your `add-spartan7-support` dev branch it seems that the `counter_ax7050.xdc` is not present in the `common` directory.

MidsummerNight, 2022-01-31:
@acomodi Hi, but it's right [here](https://github.com/MidsummerNight/symbiflow-arch-defs/blob/673568f829d79d4971698f4d99cc08eacf3e7830/xc/xc7/tests/common/counter_ax7050.xdc). Anyway I made the PR #2374 and let's see what needs to be done!

bl0x, 2023-02-24:
@MidsummerNight I've got a board with an xc7s25 and could in principle help testing / adding to the support. However, I haven't worked with all the things yet, but I'm eager to learn =)

MidsummerNight, 2023-02-27:
@bl0x Hi there, I haven't been active for the last year and thus can't recall all the details, so sorry for that. When I tried contributing to the project, the overall flow was:

1. Obtain data of Xilinx 7-series parts via [Project X-Ray](https://github.com/f4pga/prjxray)
2. Put the data obtained from Project X-Ray through the Architecture Definition (this repository) workflow. Ideally, in the end we should get a file named  `arch.timing.xml` (architecture definition file), which as the name suggests, describes the architecture of the desired part in a way that the implementation tools (VPR was the target at the time) could understand, and now we have an open source toolchain for our chip.

Unfortunately I didn't succeed in generating the arch-def file for xc7s50. As for your case, well prjxray only supports Vivado 2017.2 so far, but 2017.2 doesn't support the xc7s25 chip you have. This means you cannot even start the arch-defs workflow for xc7s25 (since it requires data from prjxray in the first place). To the extent of my knowledge, 2 solutions lie ahead:

1. Improve Project X-Ray so that it can use new Vivado versions (that support xc7s25). X-Ray is currently stuck to 2017.2 because of this [issue](https://github.com/f4pga/prjxray/issues/14). If you can help fix that, you should be able to prceed with arch-defs.
2. For whatever reason X-Ray seems to have been in hiatus for a very long time. I think the devs are shifting focus to [FPGA Interchange Formats](https://github.com/chipsalliance/fpga-interchange-tests). Never dived deep into this, but it seems like a method to extract the chips' architecture details with the help of a tool called [RapidWright](https://github.com/Xilinx/RapidWright/). Again I'm not clear on the details myself, though [this](https://github.com/f4pga/prjxray/issues/1789#issuecomment-986960472) might be useful.

MaksRawski, 2023-12-27:
Didn't [#2658](https://github.com/f4pga/f4pga-arch-defs/pull/2658) do exactly that? If so I think this issue could be closed.