f4pga/f4pga-arch-defs #2463: VPR fails to place long carry chains on xc7

kazkojima, 2022-03-01:
Hi,

I've added 256 to the bit-width parameter list of tests/7-carry_stress test

```
diff --git a/tests/7-carry_stress/CMakeLists.txt b/tests/7-carry_stress/CMakeLists.txt
index 1995e5d0..969fd84a 100644
--- a/tests/7-carry_stress/CMakeLists.txt
+++ b/tests/7-carry_stress/CMakeLists.txt
@@ -1,6 +1,6 @@
 
 get_target_property_required(PYTHON3 env PYTHON3)
-foreach(CARRY_DEPTH 1 2 3 4 7 8 9 15 16 17 31 32)
+foreach(CARRY_DEPTH 1 2 3 4 7 8 9 15 16 17 31 32 256)
     set(CARRY_TEST carry_test_${CARRY_DEPTH}_init0.v)
     add_custom_command(
         OUTPUT ${CARRY_TEST}
```

and run "make 7-carry_stress_init_fabric_depth_256_dummy_artix7_xc7a50t-basys3_test_route".
 The placement failed with the following message.

```
File: /home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/vtr-optimized_1642201887903/work/vpr/src/place/initial_placement.cpp
Line: 413
Message: Initial placement failed.
Could not place macro length 64 with head block $auto$alumacc.cc:485:replace_alu$1426.genblk1.slice[0].genblk1.carry4 (#0); not enough free locations of type(s) {BLK-TL-CLBLL_L, BLK-TL-CLBLL_R, BLK-TL-CLBLM_L, BLK-TL-CLBLM_R}.
Please manually size the FPGA because VPR can't do this yet.
```

kazkojima, 2022-03-01:
FYI, nextpnr-xilinx has a similar issue
https://github.com/gatecat/nextpnr-xilinx/issues/34

programmerjake, 2022-03-11:
Copying a comment from the nextpnr-xilinx issue that applies to VPR too:
Luke (the person at Libre-SOC who originally encountered this bug) thinks this bug should be fixed by making yosys split long carry4 chains. imho vpr and nextpnr-xilinx are where that would be handled since they know the details of how many carry4 blocks are available and should be able to insert the necessary routing between them. imho yosys is the wrong place to try to fix that.

mithro, 2022-03-11:
The easiest **_short term fix_** is for Yosys to split long carry4 chains. 

As you point out this is probably **_not_** the best place to do this however and the place and route tooling has a much better idea of the constraints around placement and hence when the carry can and can't be used. However, that is probably going to require quite a bit of work (at least it will in vpr -- don't know about nextpnr) so the short term fix is probably worth still landing.