f4pga/f4pga-arch-defs #2584: Need help to verify arch.*.xml and rr_graph.bin on spartan7

KKtiandao, 2022-04-13:
To add spartan7 into arch_def flow, I do some modification based [this pr](https://github.com/SymbiFlow/f4pga-arch-defs/pull/2374/files) and ran the arch_defs flow completely.  But how do i verify the output file like arch.*.xml and rr_graph? Simulation?

KKtiandao, 2022-04-15:
@mithro Could you give me some guidances?

acomodi, 2022-04-15:
@KKtiandao There is no validation of the arch and RR graph apart from the internal checks within VTR.

If the flow completes and you are able to get a bitstream (or at least a FASM) as an output, I believe it is enough.

What can be verified is whether the output FASM file is actually correct, running [`fasm2bels`](https://github.com/SymbiFlow/f4pga-xc-fasm2bels), which allows to decompile the FASM file into the original netlist, and verifying that everything was correctly generated by the flow.

KKtiandao, 2022-04-18:
@acomodi Thanks very much! I'll try fasm2bels.

KKtiandao, 2022-04-18:
@acomodi I see tile-types listed in the [CMakeList.txt](https://github.com/KKtiandao/f4pga-arch-defs/blob/23fc6940e14cb949b53ed3436cbe4d4014ec0fba/xc/xc7/archs/spartan7/devices/xc7s50-virt/CMakeLists.txt) is less than which acture in spartan7/artix7; While I try to append some new tile and pbtypes according to the info of prjxray-db, make failed.  How do I correct the [CMakeList.txt](https://github.com/KKtiandao/f4pga-arch-defs/blob/23fc6940e14cb949b53ed3436cbe4d4014ec0fba/xc/xc7/archs/spartan7/devices/xc7s50-virt/CMakeLists.txt)?

acomodi, 2022-04-19:
Hi @KKtiandao. Can you please open a PR with the current WIP changes you have? This can help understanding a bit more what is going on. Also the error output is quite useful to get a picture of the problem.

In general, those tiles and pb_types correspond to the ones defined [here](https://github.com/SymbiFlow/f4pga-arch-defs/tree/main/xc/xc7/archs/artix7/tiles) (for artix7) and if you want to add some other ones, e.g. the DSP which is currently not present, you need to first define the DSP tile, as well as the DSP pb_type, alongside with the other pb_types [here](https://github.com/SymbiFlow/f4pga-arch-defs/tree/main/xc/common/primitives).

KKtiandao, 2022-04-19:
> Can you please open a PR with the current WIP changes you have? This can help understanding a bit more what is going on. Also the error output is quite useful to get a picture of the problem.
@[acomodi](https://github.com/acomodi) Thanks for your reminder. I have create a [PR](https://github.com/SymbiFlow/f4pga-arch-defs/pull/2586). 

> In general, those tiles and pb_types correspond to the ones defined [here](https://github.com/SymbiFlow/f4pga-arch-defs/tree/main/xc/xc7/archs/artix7/tiles) (for artix7) and if you want to add some [...]
@acomodi  As dsp is a usefull model and it has already in the prjxray-db, why it is not present? Is there any differences  between dsp and other tiles presented?