f4pga/f4pga-arch-defs #2620: Suporting u+ is in plan?

the-centry, 2022-04-29:
If there is a plan to support u+ device?
Thanks so much!

mithro, 2022-04-29:
See https://github.com/f4pga/prjuray - https://github.com/SymbiFlow/f4pga-arch-defs/pull/1651

mithro, 2022-04-29:
Support for Ultrascale and Ultrascale+ will most likely end up happening through the FPGA Interchange Format rather than this architecture definitions approach.

You can see some initial xczu7ev support in that method -> https://chipsalliance.github.io/fpga-interchange-tests/#xczu7ev

the-centry, 2022-04-30:
> Support for Ultrascale and Ultrascale+ will most likely end up happening through the FPGA Interchange Format rather than this architecture definitions approach.
> 
> You can see some initial xczu7ev support in that method -> https://chipsalliance.github.io/fpga-interchange-tests/#xczu7ev
> 
> > **[FPGA interchange tests — fpga-interchange-tests 0.1 documentation](https://chipsalliance.github.io/fpga-interchange-tests/#xczu7ev)**

Ok,thanks so much!
This project [fpga-interchange](https://github.com/chipsalliance/python-fpga-interchange/tree/c12aff880fce39330f059f644cd4911548699272) seems like that next-pnr generate *bba through rapidwright,and the use next-pnr place and route! It may meet problem while device's data over the limit of capnp!
It means that the work generating vpr's device data through rapidwright for Ultrascale+ is in plan?