f4pga/f4pga-arch-defs #2655: "latest" files in GCS buckets no longer updated

Xiretza, 2022-05-14:
The "latest" files in the GCS bucket, e.g. https://storage.googleapis.com/symbiflow-arch-defs-gha/symbiflow-toolchain-latest point to an old version of the toolchain (currently build 597 of fb1b251a from 2022-03-22), which makes it impossible for me to download a recent archive - looking at the GCS bucket would require owning and logging into a google account. Something in the CI pipeline should probably update those files when a new archive is pushed.

umarcor, 2022-05-18:
Updating the "latest" pointers was broken during the transition from Kokoro to custom GitHub Actions runners, and renaming the default branch from `master` to `main`. FTR, it was reported in the slack channel last week. 

We are working on it:

- https://github.com/SymbiFlow/f4pga-arch-defs/commit/6529d853ff8540ab7a849f28cd68fe2d86fca209
- https://github.com/SymbiFlow/f4pga-arch-defs/pull/2665

Unfortunately, CI in this repo takes ~10h. Therefore, it is tedious to have these changes tested.

Xiretza, 2022-05-29:
#2665 was merged 6 days ago, the latest links are still not being updated. What's the current status here? Maybe the links themselves have changed with the Symbiflow->F4PGA rename?

umarcor, 2022-07-27:
@Xiretza, there is some issue with credentials and it's taking longer than expected to restore the procedure that updates the GCS buckets. Meanwhile, we implemented an alternative using GitHub Releases. See https://github.com/SymbiFlow/f4pga-arch-defs/releases/tag/latest.

We will update and close this issue when GCS buckets are updated again. Then, you can decide to keep using the Releases or go back to using the GCS urls.