f4pga/f4pga-arch-defs #2680: How to writer a new pb_model?

the-centry, 2022-05-20:
The pb_model write by experience or get information from vivado such as [a_srl.pb_type](https://github.com/SymbiFlow/f4pga-arch-defs/blob/main/xc/common/primitives/slicem/srl/a_srl.pb_type.xml).  
I want to write a cle's model.xml and pb_type.xml.
Thanks so much!

mkurc-ant, 2022-06-14:
Hi! Could you elaborate more on what you want to achieve? You are saying you want to add a model and pb_type for `cle`. That comes from the UltraScale architecture which isn't currently supported.

the-centry, 2022-06-16:
> 

I want to combine prjuray and symbiflow to expand the support devices to UltraScale+. Early I think it's possible to  change the primitive of xc to UltraScale+.But I found that the two archs are so different, So I want to create primitive for UltraScale+.However I don't know how to writer a primitive for UltraScale+. Now I try to grope the principle of xc's primitives creating!

closed 2023-05-30 by the-centry