f4pga/f4pga-arch-defs #2979: MMCM for Zynq

darktjm, 2022-10-20:
Could you please add MMCM to the Zynq7 parts?  That's half the PLLs on the device, and the more accurate ones, at that.  I'd try to do it myself, given that it appears to be nothing more than copying cmake config from the artix7, but I can't get this mess to even start building.

     Failed to find matching architecture model for 'MMCME2_ADV_VPR'

mithro, 2022-10-20:
@mkurc-ant

mkurc-ant, 2022-10-20:
@darktjm Could you share the whole log of the failing build? It would be best to share the code with your changes as well.

darktjm, 2022-10-20:
I don't see how an entire log would help, so no, I won't post it.  If you just grep for MMCM in the arch defs, you can see that it's not there for the zynq, even though the zynq clearly has them.  If you think I'm wrong, correct me and close this issue.  Surely you guys already have some sort of tests to verify that what you add actually works, yes?   If you really, really need to see the code I was trying to get working, it's at https://bitbucket.org/darktjm/ebaz.  The MMCM instantiation is done by xc7-pll.c, with the -m option.  Maybe it's wrong, but it appears to work with Vivado, and since Vivado tells me what the clock frequency is, I can tell that it's "correct", although the code which uses it (the demo subdir of the same bitbucket page) doesn't and I have no idea why (because I basically don't know what I'm doing, and I spend more time fighting tools than actually doing anything productive).   Are you going to help me debug why my demo won't produce HDMI video?  If not, I don't see any reason to post it.  In fact, since it's there on bitbucket, I guess there's no reason to post it, in any case, unless I decide to pull it.

I don't know what you mean by "my changes".  Code with my changes?  You mean where I just copied anything mentioning MMCM from the artix7 to the zynq7 (either files or partial files), as I described?  I don't see how that could help, since it's completely obvious.  It took me less time to do that than it took to search for MMCM-related issues on this site (or compose this message, for that matter).  Since I don't know what I'm doing, I probably missed something, anyway, so what's the point?  Surely whoever wrote that in the first place is more qualified to make the correct changes.  I can't build f4pga-arch-defs, not because of my changes, but because f4pga is a convoluted mess that gives me headaches, and my remaining life is too short for me to deal with that crap.  It never even gets to my changes.  It never even starts.  I might've had the patience for that sort of thing 30 years ago, but not any more.